jopohl/urh #842: How do we run the program once installed?  There is no icon in the menu.

LucidEye, 2021-02-28:
I've noticed this issue on the comments in the software installer.  Several people report the same problem I am having.
I installed the package version of the software that was available in the software app (I'm running Ubuntu MATE 20.04).
Once installed, there is no icon anywhere to run the app.  Tried uninstalling and reinstalling, and rebooting. Nothing.
How are we supposed to run this app once it's installed? Is there supposed to be an icon in the menu? Or is this supposed to be run from the command line?  There really isn't any instructions on how to do that anywhere on your github page.

jopohl, 2021-02-28:
Interesting, URH is now in the Ubuntu repos. :) Do you know whether it uses the [snap package](https://snapcraft.io/urh) for this or they created a separate package? Be it how it may to address your question, how about this:

1. Fire up your favorite terminal emulator
2. Type ``` urh ``` and hit Enter
3. Profit

LucidEye, 2021-02-28:
Installing from the 'software app' seems to install the snap package rather than the usual deb package... which is probably why it's not installing menu icons :-/

Running urh from a terminal returns this error: Gtk-Message: 09:15:27.945: Failed to load module "atk-bridge"

The app does eventually start, but it takes several seconds, it's almost like the machine hangs for a while as the app tries to start which seems like something is wrong.

Do I need to install atk-bridge? If so which package, there are 3....

Package libatk-bridge-2.0-0-udeb
buster (stable) (debian-installer): AT-SPI 2 toolkit bridge - shared library for d-i
2.30.0-5: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x

Package libatk-bridge2.0-0
buster (stable) (libs): AT-SPI 2 toolkit bridge - shared library
2.30.0-5: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x

Package libatk-bridge2.0-dev
buster (stable) (libdevel): Development files for the AT-SPI 2 toolkit bridge
2.30.0-5: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x

 
 

   
   - Type urh and hit Enter
   - Profit

jopohl, 2021-02-28:
I see, thank you for the clarification. Seems like they changed something in snap (again!). 

I guess it will not help if you install an atk-bridge package on your machine, because snap packages bundle all dependencies in them. So the snap would need to be rebuilt.
However, I recently moved to another country and left my PC with the snap building stuff back in Germany. So do not expect an update here soon. 

For the startup time: I am not sure if it is caused by the missing lib or if this is a more general problem to snap. Do you use other snaps? Do they start faster?

If you do not insist on installing URH from the Software Center I would suggest to remove it through it and then use pip to install URH natively. That should give you a way better startup time:
```bash
sudo python3 -m pip install --upgrade pip  # Update your pip installation
sudo python3 -m pip install urh            # Install URH
urh # start URH
```

PR #847 (2021-03-06): add desktop entry to snap
fix #842
closed 2021-03-06 by jopohl