jopohl/urh #845: rfcat not sending

BlessYou656, 2021-03-05:
Hi,i have problem with send data via rfcat.Rfcat now use bytes instead of string to send data..Here is error:

TypeError                                 Traceback (most recent call last)
/usr/local/lib/python3.9/dist-packages/rflib/__init__.py in <module>
----> 1 d.RFxmit('\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\')

/usr/local/lib/python3.9/dist-packages/rflib/chipcon_nic.py in RFxmit(self, data, repeat, offset)
   1125         waitlen += repeat * (len(data) - offset)
   1126         wait = USB_TX_WAIT * ((old_div(waitlen, RF_MAX_TX_BLOCK)) + 1)
-> 1127         self.send(APP_NIC, NIC_XMIT, b"%s" % struct.pack("<HHH",len(data),repeat,offset)+data, wait=wait)
   1128 
   1129     def RFxmitLong(self, data, doencoding=True):

TypeError: can't concat str to bytes

sending command shoul be: d.RFxmit(b'\\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa\xaa')

Thanks for help

andynoack, 2021-03-05:
Please try the rfcat_fix branch. Currently I do not have the rfcat environment configured, so this is just the fix you suggested. Can you confirm?

BlessYou656, 2021-03-05:
Now its working thanks

PR #846 (2022-02-21): add b before data string
Fix #845
closed 2021-03-05 by BlessYou656