jopohl/urh #848: Unable to open *.raw

blackwiz4rd, 2021-03-11:
##### Expected Behavior
Open  *.raw file without exception.

##### Actual Behavior
Traceback (most recent call last):
File "/usr/lib/python3.9/site-packages/urh/controller/MainController.py", line 564, in on_open_recent_action_triggered
self.add_files(FileOperator.uncompress_archives([action.data()], QDir.tempPath()))
File "/usr/lib/python3.9/site-packages/urh/controller/MainController.py", line 408, in add_files
self.add_signalfile(filename, group_id, enforce_sample_rate=enforce_sample_rate)
File "/usr/lib/python3.9/site-packages/urh/controller/MainController.py", line 285, in add_signalfile
signal = Signal(filename, sig_name, sample_rate=sample_rate)
File "/usr/lib/python3.9/site-packages/urh/signalprocessing/Signal.py", line 75, in __init__
self.__load_complex_file(filename)
File "/usr/lib/python3.9/site-packages/urh/signalprocessing/Signal.py", line 83, in __load_complex_file
self.iq_array = IQArray.from_file(filename)
File "/usr/lib/python3.9/site-packages/urh/signalprocessing/IQArray.py", line 203, in from_file
return IQArray(data=np.fromfile(filename, dtype=np.float32))
File "/usr/lib/python3.9/site-packages/urh/signalprocessing/IQArray.py", line 17, in __init__
self.__data = self.convert_array_to_iq(data)
File "/usr/lib/python3.9/site-packages/urh/signalprocessing/IQArray.py", line 212, in convert_array_to_iq
return arr.reshape((-1, 2), order="C")
ValueError: cannot reshape array of size 12717211 into shape (2)

##### Steps To Reproduce

1. Save file to file sink *.raw in gnuradio
2. Open file in urh

##### Platform Specifications

 - OS: Arch Linux
 - URH version: 2.9.1-1
 - Python version: 3.9.2
 - Installed via pacman

jopohl, 2021-03-11:
What complex format has the .raw file? 2xint8, 2xuint8, 2xint16, ...? Rename it accordingly, see https://github.com/jopohl/urh/wiki/Supported-signal-file-formats

blackwiz4rd, 2021-03-13:
It is a Float 32 in gnuradio, so it should be `.complex` in urh. After capture sometimes it gets opened, other times not.

jopohl, 2021-03-13:
> After capture sometimes it gets opened, other times not.

This sounds rather like a bug in GNU Radio and/or a problem with your flow-graph. This error indicates that the file gets not completely written by the application.

jopohl, 2021-03-13:
As a side question: Why do you capture in GR and not directly in URH?

blackwiz4rd, 2021-03-13:
> > After capture sometimes it gets opened, other times not.
> 
> This sounds rather like a bug in GNU Radio and/or a problem with your flow-graph. This error indicates that the file gets not completely written by the application.

I thought so too, however I can open it with a simple `np.fromfile(filename, np.float32)`.

> As a side question: Why do you capture in GR and not directly in URH?

That is because I need to set up more options during capture (e.g. enable direct sampling mode, set RF gain) and also add filters if necessary.

jopohl, 2021-03-13:
> I thought so too, however I can open and display it with a simple np.fromfile(filename, np.float32)

That only gives you a stream of float32, not complex. Try `np.fromfile(filename, np.complex64)`, if **that** works for a file in a Python shell that makes URH crash we have indeed a bug. Otherwise GNU Radio swallowed half of the last sample.

blackwiz4rd, 2021-03-13:
> That only gives you a stream of float32, not complex. Try `np.fromfile(filename, np.complex64)`, if **that** works for a file in a Python shell that makes URH crash we have indeed a bug. Otherwise [...]

I have tried with `np.complex64` and it works correctly. I have sent you an email with one file that won't open in urh.
Thanks.

jopohl, 2021-03-13:
Thank you for providing the file! Seems like numpy just skips the last sample to avoid a conversion error. I implemented the same behavior in #849.

commit 71153bab (2021-03-13): allow incomplete complex files to load fix #848
closed 2021-03-13 by jopohl