jopohl/urh #852: Import WAV as demodulated signal

heeplr, 2021-03-21:
I got [this signal](https://gist.github.com/heeplr/259647d574072c2b7d95ebdf5a9d24b9/raw/dcb130377defaf70ebe1468b51718723665c4403/ttl.wav) (WAV) which doesn't have any modulation.

I'm not sure if importing it as demodulated signal would really help or even if those kind of signals are within the scope of urh, but it would be nice if they could be analyzed with urh.

The signal seems to be FSK with one pulse per bit. Short pulses = 0, Long pulses = 1.
First there's a preamble with some 1's, then it seems every 8 bit are followed by 2 stop bits.

andynoack, 2021-03-22:
Actually your signal looks like FSK before demodulation...

heeplr, 2021-03-27:
Is it possible to demodulate this with URH? I don't seem to get reliable results (i.e. the preamble is mixed 0 and 1). 
One Symbol seems to be ~20 samples since it's only one period. I tried with various settings from 2 to 130 samples per symbol. Can URH handle FSK modulation with 1 symbol per period?

jopohl, 2021-03-27:
How did you capture this signal? URH uses instantaneous frequencies to demodulate FSK which relies on IQ signal format. Every common SDR should give you IQ data, but certainly demodulation will not work for a mono channel WAV.

heeplr, 2021-03-27:
@jopohl I recorded the signal with the soundcard input. It is a mono channel "audio" signal for inband signalling like a modem & a phone line.

jopohl, 2021-03-27:
Ah that's interesting! URH is not really made to demodulate such a signal. It rather relies on the input from a SDR, capturing with the soundcard is an interesting use-case. What happens when you load a mono-channel WAV is, it will interpret it as "already demodulated" (i.e. should be a rectangular signal). Never thought of your use-case to be honest.

heeplr, 2021-03-27:
Thank you for clarifying.
Well, a lot of interesting stuff is happening in the audible spectrum :)
Honestly I thought it would be straightforward. URH is wonderful to analyze protocols so it was my first choice.
Mono WAVs being interpreted as "already demodulated" is exactly why I initially opened the issue (and thought would help). The WAVs I import offer "Signal View:" "Analog", "Demodulated" and "Spectrum" but the demodulated view differs from the analog one (and it's wrongly demodulated).
When I open the signal in an audio editor (audacity) it looks like the analog view in URH.

jopohl, 2021-03-27:
It is not as straightforward as it appears at first sight. :) 

URH is completely built upon IQ demodulation, therefore, it treats mono channel wavs as "already demodulated" because it has no way to demodulate it properly.
Having said that, it is certainly possible to demodulate the signal you provided. The challenge is to get it into URH without changing the whole demodulation framework.
If we do some math we might be able to transform the single channel wave into a complex format (read: double channel wave) and use URH's existing demodulation methods. Let me reopen this issue and change the title a bit.

mir123, 2023-04-28:
This would also be fantastic for all of the acoustic (i.e. underwater) data protocols out there