jopohl/urh #859: URH Version 2.9.2 crashes always at startup - OSX 10.13.6

Bigblock, 2021-04-17:
Process:               urh [655]
Path:                  /Applications/urh.app/Contents/MacOS/urh
Identifier:            urh
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        urh [653]
Responsible:           urh [655]
User ID:               501

Date/Time:             2021-04-17 19:10:10.052 +0200
OS Version:            Mac OS X 10.13.6 (17G14042)
Report Version:        12
Anonymous UUID:        B42B6793-13C1-8123-12A2-8E56085CE63E

Time Awake Since Boot: 560 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x10:
--> 
    __TEXT                 0000000100d7c000-0000000100d84000 [   32K] r-x/r-x SM=COW  /Applications/urh.app/Contents/MacOS/urh

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff2b693769 CFBundleCopyBundleURL + 9
1   QtCore                        	0x0000000105559470 QLibraryInfo::location(QLibraryInfo::LibraryLocation) + 1328
2   QtCore                        	0x00000001055596cd QLibraryInfo::location(QLibraryInfo::LibraryLocation) + 1933
3   QtCore                        	0x000000010573f494 QCoreApplication::libraryPathsLocked() + 1524
4   QtCore                        	0x000000010573b57b QCoreApplication::libraryPaths() + 43
5   QtCore                        	0x000000010572a0a4 QFactoryLoader::update() + 36
6   QtCore                        	0x000000010572b9c0 QFactoryLoader::QFactoryLoader(char const*, QString const&, Qt::CaseSensitivity) + 224
7   QtGui                         	0x0000000105fd1b3c QGlobalStatic<QFactoryLoader, (anonymous namespace)::Q_QGS_loader::innerFunction(), (anonymous namespace)::Q_QGS_loader::guard>::operator()() + 172
8   QtGui                         	0x0000000105fd1ef1 QPlatformIntegrationFactory::keys(QString const&) + 785
9   QtGui                         	0x0000000105fdee5a QGuiApplicationPrivate::createPlatformIntegration() + 1690
10  QtGui                         	0x0000000105fe03fb QGuiApplicationPrivate::createEventDispatcher() + 27
11  QtCore                        	0x000000010573b1af QCoreApplicationPrivate::init() + 1599
12  QtGui                         	0x0000000105fdab59 QGuiApplicationPrivate::init() + 57
13  QtWidgets                     	0x0000000107157ada QApplicationPrivate::init() + 26
14  QtWidgets.abi3.so             	0x00000001067fd37b init_type_QApplication(_sipSimpleWrapper*, _object*, _object*, _object**, _object**, _object**) + 171
15  sip.cpython-37m-darwin.so     	0x0000000105bf99cb sipSimpleWrapper_init + 203
16  libpython3.7m.dylib           	0x00000001041bac1b type_call + 235
17  libpython3.7m.dylib           	0x0000000104130f43 _PyObject_FastCallKeywords + 179
18  libpython3.7m.dylib           	0x00000001042667c3 call_function + 515
19  libpython3.7m.dylib           	0x000000010425f0f4 _PyEval_EvalFrameDefault + 23764
20  libpython3.7m.dylib           	0x000000010413143e _PyFunction_FastCallKeywords + 206
21  libpython3.7m.dylib           	0x000000010426667c call_function + 188
22  libpython3.7m.dylib           	0x000000010425f0f4 _PyEval_EvalFrameDefault + 23764
23  libpython3.7m.dylib           	0x0000000104257ccf _PyEval_EvalCodeWithName + 495
24  libpython3.7m.dylib           	0x0000000104257ad0 PyEval_EvalCode + 48
25  urh                           	0x0000000100d7e152 0x100d7c000 + 8530
26  urh                           	0x0000000100d7e54e 0x100d7c000 + 9550
27  urh                           	0x0000000100d7ca14 0x100d7c000 + 2580

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff539d9bdc start_wqthread + 0
1   ???                           	0x0000000052ab8c01 0 + 1386974209

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff539d9bdc start_wqthread + 0
1   ???                           	0x000000ef01000000 0 + 1026513960960

[... 340 lines cut ...]
mapped file                       25.5M        2 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                            843.9M      762 
TOTAL, minus reserved VM space   811.2M      762 

Model: MacBookPro8,3, BootROM 87.0.0.0.0, 4 processors, Intel Core i7, 2,5 GHz, 16 GB, SMC 1.70f5
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x85F7, 0x483634314755363746393333334700000000
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x85F7, 0x483634314755363746393333334700000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 6.0.7f22, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 840 EVO 500GB, 500,11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple Internal Keyboard / Trackpad
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

Bigblock, 2021-04-17:
OK, gerade gesehen, dass OSX 10.14 die Mindestversion ist. Bis 2.9.1 hat URH bei mir funktioniert. Jetzt kann ich auch ältere Versionen nicht mehr starten.

jopohl, 2021-04-17:
Is OSX able to run docker? In that case you might run the [URH docker image](https://hub.docker.com/r/jopohl/urh/)

jopohl, 2021-04-17:
Alternatively, installing URH with pip should also work for you, see [here](https://github.com/jopohl/urh#installation) (click macOS -> With pip)

Bigblock, 2021-04-18:
Got OSX 10.15 running on my Mac now. URH 2.9.1 is installable now, but URH 2.9.2 crashes with this message:

Process:               urh [3051]
Path:                  /Applications/urh.app/Contents/MacOS/urh
Identifier:            urh
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        urh [3050]
Responsible:           urh [3050]
User ID:               501

Date/Time:             2021-04-18 02:13:31.314 +0200
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Anonymous UUID:        5ED8AED2-AD72-F6C1-A5DE-26B740B23C35

Time Awake Since Boot: 2500 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [3051]

VM Regions Near 0x10:
--> 
    __TEXT                 0000000107264000-000000010726c000 [   32K] r-x/r-x SM=COW  /Applications/urh.app/Contents/MacOS/urh

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff325bbcf8 CFBundleCopyBundleURL + 9
1   QtCore                        	0x000000010bee9470 QLibraryInfo::location(QLibraryInfo::LibraryLocation) + 1328
2   QtCore                        	0x000000010bee96cd QLibraryInfo::location(QLibraryInfo::LibraryLocation) + 1933
3   QtCore                        	0x000000010c0cf494 QCoreApplication::libraryPathsLocked() + 1524
4   QtCore                        	0x000000010c0cb57b QCoreApplication::libraryPaths() + 43
5   QtCore                        	0x000000010c0ba0a4 QFactoryLoader::update() + 36
6   QtCore                        	0x000000010c0bb9c0 QFactoryLoader::QFactoryLoader(char const*, QString const&, Qt::CaseSensitivity) + 224
7   QtGui                         	0x000000010f500b3c QGlobalStatic<QFactoryLoader, (anonymous namespace)::Q_QGS_loader::innerFunction(), (anonymous namespace)::Q_QGS_loader::guard>::operator()() + 172
8   QtGui                         	0x000000010f500ef1 QPlatformIntegrationFactory::keys(QString const&) + 785
9   QtGui                         	0x000000010f50de5a QGuiApplicationPrivate::createPlatformIntegration() + 1690
10  QtGui                         	0x000000010f50f3fb QGuiApplicationPrivate::createEventDispatcher() + 27
11  QtCore                        	0x000000010c0cb1af QCoreApplicationPrivate::init() + 1599
12  QtGui                         	0x000000010f509b59 QGuiApplicationPrivate::init() + 57
13  QtWidgets                     	0x0000000110163ada QApplicationPrivate::init() + 26
14  QtWidgets.abi3.so             	0x000000010fd2c37b init_type_QApplication(_sipSimpleWrapper*, _object*, _object*, _object**, _object**, _object**) + 171
15  sip.cpython-37m-darwin.so     	0x000000010c5d49cb sipSimpleWrapper_init + 203
16  libpython3.7m.dylib           	0x000000010ad31c1b type_call + 235
17  libpython3.7m.dylib           	0x000000010aca7f43 _PyObject_FastCallKeywords + 179
18  libpython3.7m.dylib           	0x000000010addd7c3 call_function + 515
19  libpython3.7m.dylib           	0x000000010add60f4 _PyEval_EvalFrameDefault + 23764
20  libpython3.7m.dylib           	0x000000010aca843e _PyFunction_FastCallKeywords + 206
21  libpython3.7m.dylib           	0x000000010addd67c call_function + 188
22  libpython3.7m.dylib           	0x000000010add60f4 _PyEval_EvalFrameDefault + 23764
23  libpython3.7m.dylib           	0x000000010adceccf _PyEval_EvalCodeWithName + 495
24  libpython3.7m.dylib           	0x000000010adcead0 PyEval_EvalCode + 48
25  urh                           	0x0000000107266152 0x107264000 + 8530
26  urh                           	0x000000010726654e 0x107264000 + 9550
27  urh                           	0x0000000107264a14 0x107264000 + 2580

Thread 1:
0   libsystem_pthread.dylib       	0x00007fff6c93fb68 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	0x00007fff6c93fb68 start_wqthread + 0

[... 367 lines cut ...]
mapped file                       27.3M        2 
shared memory                       48K        6 
===========                     =======  ======= 
TOTAL                              1.1G      890 
TOTAL, minus reserved VM space     1.0G      890 

Model: MacBookPro8,3, BootROM 87.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 2,5 GHz, 16 GB, SMC 1.70f5
Graphics: kHW_IntelHD3000Item, Intel HD Graphics 3000, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x85F7, 0x483634314755363746393333334700000000
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x85F7, 0x483634314755363746393333334700000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.102.31)
Bluetooth: Version 7.0.6f7, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 840 EVO 500GB, 500,11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

jopohl, 2021-04-18:
Is the same happening when you install URH with pip?

Bigblock, 2021-04-18:
If i try to install via pip it also doesn't work. Error message is:

`xxxs-MBP:urh xxx$ python setup.py install
  File "setup.py", line 44
    file=sys.stderr)
        ^
`
python Setuptools are installed correctly asfaik:
Requirement already satisfied: setuptools in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (49.2.1)

jopohl, 2021-04-18:
This log output is pretty cropped, but I would guess you are using python2 here. What does 

```bash
python --version
which python
which python3
```

give you?

Bigblock, 2021-04-19:
phyton wasn't linked correctly.

After installing python 3.9.4 and linking it correctly in OSX 10.15.7 i've installed URH from source.

While installation i've got this error messages:

Markuss-MBP:~ markus$ python --version
Python 3.9.4
Markuss-MBP:~ markus$ cd urh
Markuss-MBP:urh markus$ python setup.py install
Skipping native support for airspy
Skipping native support for bladerf
Skipping native support for hackrf
Skipping native support for limesdr
Skipping native support for plutosdr
Found rtlsdr lib. Will compile with native rtlsdr support
Skipping native support for usrp
Skipping native support for sdrplay
Compiling src/urh/cythonext/awre_util.pyx because it depends on /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urh-2.9.2-py3.9-macosx-10.9-x86_64.egg/urh/cythonext/util.pxd.
Compiling src/urh/cythonext/signal_functions.pyx because it depends on /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urh-2.9.2-py3.9-macosx-10.9-x86_64.egg/urh/cythonext/util.pxd.
Compiling src/urh/cythonext/path_creator.pyx because it depends on /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urh-2.9.2-py3.9-macosx-10.9-x86_64.egg/urh/cythonext/util.pxd.
Compiling src/urh/cythonext/auto_interpretation.pyx because it depends on /Users/markus/urh/.eggs/numpy-1.20.2-py3.9-macosx-10.9-x86_64.egg/numpy/__init__.pxd.
Compiling src/urh/cythonext/util.pyx because it depends on /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urh-2.9.2-py3.9-macosx-10.9-x86_64.egg/urh/cythonext/util.pxd.
Compiling src/urh/dev/native/lib/rtlsdr.pyx because it depends on /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urh-2.9.2-py3.9-macosx-10.9-x86_64.egg/urh/dev/native/lib/crtlsdr.pxd.
[1/6] Cythonizing src/urh/cythonext/auto_interpretation.pyx
[2/6] Cythonizing src/urh/cythonext/awre_util.pyx
[3/6] Cythonizing src/urh/cythonext/path_creator.pyx
[4/6] Cythonizing src/urh/cythonext/signal_functions.pyx
[5/6] Cythonizing src/urh/cythonext/util.pyx
[6/6] Cythonizing src/urh/dev/native/lib/rtlsdr.pyx
running install
running bdist_egg
running egg_info
writing src/urh.egg-info/PKG-INFO
writing dependency_links to src/urh.egg-info/dependency_links.txt
writing entry points to src/urh.egg-info/entry_points.txt
writing requirements to src/urh.egg-info/requires.txt
writing top-level names to src/urh.egg-info/top_level.txt
package init file 'src/urh/__pycache__/__init__.py' not found (or not a regular file)
package init file 'src/urh/dev/native/__pycache__/__init__.py' not found (or not a regular file)
package init file 'src/urh/dev/__pycache__/__init__.py' not found (or not a regular file)
Finalizing options
reading manifest file 'src/urh.egg-info/SOURCES.txt'
writing manifest file 'src/urh.egg-info/SOURCES.txt'
installing library code to build/bdist.macosx-10.9-x86_64/egg
running install_lib
running build_py
running build_ext
building 'urh.cythonext.awre_util' extension
gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch x86_64 -g -I/Library/Frameworks/Python.framework/Versions/3.9/include/python3.9 -I/Users/markus/urh/.eggs/numpy-1.20.2-py3.9-macosx-10.9-x86_64.egg/numpy/core/include -c src/urh/cythonext/awre_util.cpp -o build/temp.macosx-10.9-x86_64-3.9/src/urh/cythonext/awre_util.o  -Wno-#warnings
g++ -bundle -undefined dynamic_lookup -arch x86_64 -g build/temp.macosx-10.9-x86_64-3.9/src/urh/cythonext/awre_util.o -o build/lib.macosx-10.9-x86_64-3.9/urh/cythonext/awre_util.cpython-39-darwin.so 
building 'urh.cythonext.signal_functions' extension
gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch x86_64 -g -I/Library/Frameworks/Python.framework/Versions/3.9/include/python3.9 -I/Users/markus/urh/.eggs/numpy-1.20.2-py3.9-macosx-10.9-x86_64.egg/numpy/core/include -c src/urh/cythonext/signal_functions.cpp -o build/temp.macosx-10.9-x86_64-3.9/src/urh/cythonext/signal_functions.o  -Wno-#warnings
src/urh/cythonext/signal_functions.cpp:6035:23: warning: code will never be executed
      [-Wunreachable-code]
        if ((1 == 0)) abort();
                      ^~~~~
src/urh/cythonext/signal_functions.cpp:6035:14: note: silence by adding parentheses to
      mark code as explicitly dead
        if ((1 == 0)) abort();
             ^
             /* DISABLES CODE */ ( )
src/urh/cythonext/signal_functions.cpp:7457:23: warning: code will never be executed
      [-Wunreachable-code]
        if ((1 == 0)) abort();
                      ^~~~~
src/urh/cythonext/signal_functions.cpp:7457:14: note: silence by adding parentheses to
      mark code as explicitly dead
        if ((1 == 0)) abort();
             ^
[... 923 lines cut ...]

While trying to start URH 2.9.2 i've got this error messages:

Markuss-MBP:~ markus$ urh
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/bin/urh", line 33, in <module>
    sys.exit(load_entry_point('urh==2.9.2', 'console_scripts', 'urh')())
  File "/Library/Frameworks/Python.framework/Versions/3.9/bin/urh", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
[... cut ...]
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/urh-2.9.2-py3.9-macosx-10.9-x86_64.egg/urh/main.py", line 9, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/PyQt5-5.15.4-py3.9-macosx-10.9-x86_64.egg/PyQt5/QtCore.abi3.so, 2): Library not loaded: @rpath/QtCore.framework/Versions/5/QtCore
  Referenced from: /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/PyQt5-5.15.4-py3.9-macosx-10.9-x86_64.egg/PyQt5/QtCore.abi3.so
  Reason: image not found
Markuss-MBP:~ markus$

jopohl, 2021-04-19:
Seems like PyQt5 installation is messed up. Can you run
```
 python -c "from PyQt5.QtCore import QT_VERSION_STR; print(QT_VERSION_STR)" 
```
if not, try installing another version of PyQt5 with pip, e.g.,
```
pip install pyqt5==5.15.2
```

Bigblock, 2021-04-20:
PyQt5-5.15.4 was installed. Uninstalled it and replaced it with pyqt5-5.15.2.

This version was officially downloaded:
PyQt5-5.15.2-5.15.2-cp35.cp36.cp37.cp38.cp39-abi3-**macosx_10_13_**intel.whl

Still get
File "<string>", line 1, in <module> ImportError: No module named PyQt5.QtCore

Updated my Mac from OSX 10.13 to 10.15 to get URH 2.9 running officially. OSX is not officially supported on my Mac.

I have no idea why this happens. Does my Mac need a special PyQt5 version for OSX 10.15? Is that why something wents wrong?

jopohl, 2021-04-20:
Who knows...since this is rather a PyQt5 issue than a URH one I suggest you contact the PyQt5 developers to find out what is wrong.

As an alternative you could also try installing urh with homebrew
```
brew install urh
```
and if that doesn't work ask for help at their GitHub repo.

hashier, 2021-04-20:
For me the:
- DMG 2.9.2 doesn't work
- DMG 2.9.1 works (light theme, even though OS (10.15.7 (19H524)) is in dark mode)
- pip install urh works (2.9.2) (including correct theme)
- brew install urh works (2.9.2) (including correct theme)

```
$ python -c "from PyQt5.QtCore import QT_VERSION_STR; print(QT_VERSION_STR)"
5.15.2
```

jopohl, 2021-04-20:
Thanks for your thorough testing @hashier .

Is URH from pip still working when you upgrade to PyQt 5.15.4? Might be an issue with that version on Mac and the DMG has this one bundled. I could create a fixed DMG with a lower version if this is the root cause.

hashier, 2021-04-20:
This works:

```
(venv) /tmp $ urh
(venv) /tmp $ pip freeze
Cython==0.29.23
numpy==1.20.2
psutil==5.8.0
PyQt5==5.15.4
PyQt5-Qt5==5.15.2
PyQt5-sip==12.8.1
urh==2.9.2
```

jopohl, 2021-04-21:
Found the issue and replaced the bad DMG in the 2.9.2 release with an updated one. Should work now - just drop a comment here if not.

closed 2021-04-21 by jopohl

Bigblock, 2021-04-22:
Works. Thanks!

hashier, 2021-04-22:
For history: https://github.com/jopohl/urh/commit/a9cb440011917488f2ad90f7731aa2f3bad34a0e is the commit that fixed it. (I think)