jopohl/urh #860: Send Via rfcat not doing working

pickeditmate, 2021-04-18:
##### Expected Behavior
Send bitstream using rfcat

##### Actual Behavior
button press does nothing, no TX light on Yardstick one

##### Steps To Reproduce
enable rfcat in options
give location of rfcat
notification below says it sees rfcat
select bitstream in generator
press "send via rfcat button"

##### Screenshots

##### Platform Specifications

 - OS: Windows 10 
 - URH version: 2.9.2
 - Python version: 3.8
 - Installed via: windows installer

andynoack, 2021-04-18:
Have you tried the rfcat_fix branch?

pickeditmate, 2021-04-22:
Yeah tried last night with the branch and it didn't work, also before that I tried just adding b to the string line as was done in the branch. 

I get nothing happening when I press it, no window, confirmation nor any tx light on the yardstick.

I'm on windows 10 64 bit fully windows updated 
Python 3.9 only
Latest RFCat from atlas.

Should I get a window pop up after pressing the send via rfcat button?

meeas, 2021-08-05:
@pickeditmate, Quick question, did you verify that rfcat was working on its own before trying it in urh?  There is a known issue with the default firmware that ships with the Yardstick which makes it incompatible with Windows 10, which can only be fixed by flashing the Yardstick on laptop with Linux installed on bare metal (doesn't work in a VM).  I've been using the urh feature successfully for the last few versions of both urh and rfcat, at least for 2+ years, and have had a few hundred students using that feature in my courses.  

The easiest test is to open the `rfcat -r` shell and try sending some packets.  You can copy/paste the following code to the rfcat shell as a simple test then use urh and a SDR to capture to packets to verify rfcat functionality.

```
import time
d.setFreq(433000000)
d.setMdmModulation(MOD_ASK_OOK)
while True:
    time.sleep(0.5)
    d.RFxmit("Hello World")
```

BTW, I recommend setting your SDR gain between 0 and 10 in the urh capture window so you keep your noise floor low in your capture.

Botox200, 2021-09-01:
Might be a Python issue (rfcat has problems with the new python >3 version). 
Change the line 160 in the plugin file /usr/local/lib/python3.8/dist-packages/urh/plugins/RfCat/RfCatPlugin.py
from
prepared_data = "d.RFxmit({})".format(str(data)[11:-1]) #[11:-1] Removes "bytearray(b...)
to
prepared_data = "d.RFxmit(b{})".format(str(data)[11:-1]) #[11:-1] Removes "bytearray(b...)

This "b" will allow transmitting the code.

pickeditmate, 2021-09-03:
> @pickeditmate, Quick question, did you verify that rfcat was working on its own before trying it in urh? There is a known issue with the default firmware that ships with the Yardstick which makes it [...]

Thanks for the handy hints they'll definitely come in handy so thanks for that.

When I first for the rfcat I initially had that problem and quickly fixed it with the latest hex file and a greatfet.
It's definitely working and I've reversed a good few wireless door chimes, just for whatever reason URH gives no indication that's its sending anything and when checking with SDR and pink pager it draws a blank.

I'm just going to leave that part of URH alone for now as it's got to be something fish with my environment somewhere, I'm using purely python3 and installed rfcrack using pip3.

andynoack, 2021-10-19:
Closed due to inactivity