jopohl/urh #864: Cannot use SoundCard on MacOS Big Sur

NeuroForLunch, 2021-05-01:
##### Expected Behavior
SoundCard sampling to work

##### Actual Behavior

[image]

##### Platform Specifications
 - OS: MacOS Big Sur
 - URH version: v2.9.2
 - Python version: 3.9
 - Installed via DMG

NeuroForLunch, 2021-05-01:
Apparently this is due to the app not asking for microphone permission and I can't find a way to manually grant it.

I bypassed this problem by using 

```
pip install cython
pip install pyaudio
python -m pip install --upgrade urh
```

and now audio works but there's another problem: it won't let me set the default frequency below 100k. I can type in a lower frequency but when I press okay and reopen settings, its back to 100k.

[image]

[image]

andynoack, 2021-10-19:
Closed due to inactivity

NeuroForLunch, 2021-10-19:
The issue still remains

gramss, 2021-12-31:
Just gathering some useful information to fix this issue.

Based on the [apple docs](https://developer.apple.com/documentation/avfoundation/cameras_and_media_capture/requesting_authorization_for_media_capture_on_macos
), one need to add the property `NSMicrophoneUsageDescription` to the Info.plist of the packaged application in order to access the microphone properly.

So it would probably be good to add some properties to that Info.plist around here, to fix the DMG-Release: https://github.com/jopohl/urh/blob/a9cb440011917488f2ad90f7731aa2f3bad34a0e/data/pyinstaller_helper.py#L60-L66
For launching `urh` directly from the MacOS terminal, the terminal itself should need to be whitelisted.

I also found [this thread from the QT forum](https://forum.qt.io/topic/97570/macos-mojave-privacy-prompt-not-appearing-when-debugging-audio-app-in-qt-creator) to be relevant:

> Note that the security prompt seems to use the app name from the following Info.plist field to register the app.
> `<key>CFBundleIdentifier</key>`
> `<string>Company Name.Application Name</string>`

This [python sub-function](https://github.com/kovidgoyal/kitty/blob/072fe91518dabe6c7f8c41d030b257e817eaea53/setup.py#L986) might also be interesting for generating useful content of a Info.plist.

PR #945 (2022-02-22): Fix microphone access for macOS DMG
This fixes #864 .  Thanks to @gramss for the research work!
closed 2022-02-22 by jopohl