jopohl/urh #865: Unable to set frequency below 100 kHz

NeuroForLunch, 2021-05-01:
##### Expected Behavior
Set & save default frequency below 100 kHz

##### Actual Behavior

I can type in a lower frequency but when I press okay and reopen settings, its back to 100k.

##### Steps To Reproduce
[image]

[image]

##### Platform Specifications
OS: MacOS Big Sur
URH version: v2.9.2
Python version: 3.9
Installed via pip

andynoack, 2021-10-19:
Closed due to inactivity

NeuroForLunch, 2021-10-19:
The issue still remains