jopohl/urh #868: Question: multi-sources, multi-channels, api, passive radar

xloem, 2021-05-02:
This project seems very focused on a single signal read by a single receiver.  Is it the place for more versatile ideas, or are there better projects elsewhere?

It seems ideally the infrastructure might be migrated into blocks for gnuradio or pothos so that more complex prototyping can easily done.

andynoack, 2021-05-04:
Well, URH is designed to handle multiple signals at once (depends of course on your amount of RAM). Unless you are not planning such things like batch processing, you can use URHs default workflow.
If you are planning to use batch processing and similar, you can have a look at the protocol sniffer or the simulator component.

Migrating some URH blocks to GnuRadio is rather unlikely, because as far as I know, Gnuradio is written in C++ and URH in Python3.

xloem, 2021-05-04:
I'll have to look deeper for the multi-signal usecases.

Gnuradio is all about python: https://www.gnuradio.org/doc/doxygen-3.7.6.2/page_python_blocks.html

closed 2021-05-16 by jopohl