jopohl/urh #869: Possible 2 bit/symbal mapping issue with 4-FSK from rfcat/yardstick

meeas, 2021-05-08:
This is the first time I've played with 4-FSK, but it seems there is a mapping difference between the four frequencies and the 2-bit values they should represent.  Is there a way to change mappings in URH?

Here is the 4-FSK capture I'm trying to decode, greatly cropped to a single transmission for ease of download:

https://drive.google.com/file/d/11-dvKAze6dalqrhC9p5yqVTSf_nl60iM/view?usp=sharing

All the capture settings should be in the file name.  It was just a simple string "Hello World" sent at 433MHz with 4-FSK with the default 4-byte preamble and 2-byte sync word of 0x0C4E.  

It seems the preamble and sync word is being sent with 2-FSK (decodes fine with 1 bit/symbol) and the data at 4-FSK (can visually see four levels in FSK decode).  This seems strange, but I guess could be normal.   Like I said, this is my first time working with 4-FSK.  So after verifying the preamble and sync word at 1 bit/symbol, I changed it to 2 bits/symbol to try decoding the data.  After adjusting the levels and applying a moving average to clean it up a bit (download was saved after the moving average), I wasn't able to get the data to decode.  

Here is a screenshot of the decode and settings in URH:

https://drive.google.com/file/d/1lg-l4udXdliawMoV_C9feHBBRQGEAHUL/view?usp=sharing

After doing a hand comparison of the data's decoded bitstream (at 2 bits/symbol):

```
000110010010000000101101001011010010111101100101000000110010111100110110001011010010000101
```

I compared it to the data bitstream I expected to recieve.

```
0100100001100101011011000110110001101111001000000101011101101111011100100110110001100100
```

In Hex:  `48 65 6c 6c 6f 20 57 6f 72 6c 64`
In ASCII:  `Hello World`

Ignoring the last two garbage bits in the capture, I was able to determin all the data was present, but I just had a sybol -> 2-bit mapping issue.  

If you convert every instance of `00` a 2-bit boundarys from the capture to `01` and you convert `01` to `00`, everthing decodes just fine.  The `11` and the `10` mappings are just fine.  Basically it seems the 2-bit values for the two lowest symbols (from a urh decode perspective) are swapped.

So is this a bug in either URH or CC1101/Yardstick/rfcat?  Or does URH have a mapping feature when I can change what each 4-fsk symbol/frequency maps to?  Or perhaps a a decode function (didn't see one in list) that could fix this?  

And just a general knowledge question.  Is the mapping for 4-FSK different per chip implimentation?  And is it normal not to apply it to the sync word?

 - OS: Debian Testing
 - URH version: 2.9.2 (had same issue in 2.8 series as well)
 - Python version: 3.9.2
 - Installed via pip

jopohl, 2021-05-09:
Yes the mapping between the FSK frequencies and the symbols is indeed hardcoded. You can display the mapping btw by using the context menu on the graph and show the legend.

When wee were implementing this back in the day, our idea was that you can change the mapping, if needed, with a substitution decoding in Analysis. Did you try that?

meeas, 2021-05-12:
I did, but let me go back and try again now that I know that is the recommended path.  I'm out of the office this week, but I'll will try again when I get back in the office next week.