jopohl/urh #870: Win10 RTL-SDR - no support. GNU Radio?

bogus105, 2021-05-15:
I was trying to find issue #139 as metioned here in comments:
https://www.rtl-sdr.com/reverse-engineering-signals-universal-radio-hacker-software/comment-page-1/#comment-216211

speciffically:

> February 22, 2017 - 8:50 am
Jerome
The author tolds on the issues board of the project, that RTL-SDR is not natively supported. One must use a GnuRadio backend (see bug #139). Any volounteers is needed to add native rlt-sdr support !

Please direct me how to make URH work on my Win10 64-bit.
Thank you and thank you for the great piece of software!

jopohl, 2021-05-15:
This comment is very old, RTL-SDR works of the box on Windows natively. Just use the exe installer on the releases page: https://github.com/jopohl/urh/releases/tag/v2.9.2

closed 2021-05-15 by jopohl

bogus105, 2021-05-15:
Thank you for reply. On mine it's ont working out of the box. On my machine RTL-SDR option is grayed out. It is not enabled. Kindly see attached.
[image]

jopohl, 2021-05-15:
On the screenshot you posted RTL-SDR is enabled through native backend...

bogus105, 2021-05-15:
If so i don't understand why it's not working when i want to record any signal? Why is shows "not available"?

jopohl, 2021-05-16:
Please make a screenshot.

bogus105, 2021-05-16:
I'm getting eeror code -12:

[image]

jopohl, 2021-05-16:
So the native support **is** working. This error code comes from the RTL-SDR driver. Did you google it? Some things that come to my mind:
1. Try run URH as administrator
2. Try a different USB port
3. Not sure if this is needed for RTL-SDR but for HackRF you need to install driver with Zadig. Might be worth a shot.