jopohl/urh #874: Problem to activate RTL-SDR under Ubuntu 20.04

dwarning, 2021-05-19:
##### Expected Behavior
Open the RTL-SDR USB Stick for Recording signals

##### Actual Behavior
If I activate the spectrum analyzer or the recorder I got:
RTLSDR-OPEN (#0): Error Code: -1 (-1)
RTLSDR: failed to start rx mode
BTW: same error if I not connect the stick into USB port.
All my other rtl-sdr tools are working: GQRX, Gnu Radio or rtl_433.
blacklist dvb_usb_rtl28xxu is there and rtl-sdr.rules is in /etc/udev/rules.d

##### Platform Specifications
 - OS: Ubuntu 20.04
 - URH version: 2.9.2
 - Python version: 3.8
 - Installed via Snap Store

andynoack, 2021-05-19:
Have you tried starting as root?

dwarning, 2021-05-19:
No sorry - I haven't. How can I do it?

 >urh works

 >sudo urh doesn't work - permission problems mkdir

I installed udev rules for this suite:

https://github.com/jopohl/urh/wiki/SDR-udev-rules

But no success so far.

andynoack, 2021-05-19:
Be sure that you selected the native backend (not GR) in the device options. If there are permission problems you should run urh with sudo. Permission problems with sudo, but not without, sounds very strange.

dwarning, 2021-05-19:
Yes - native backend is activated.

***@***.***:~$ urh

Qt: Session management error: Could not open network socket
[INFO::settings.py::get_receive_buffer_size] Allocate receive buffer 
with 262,144KB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode
[ERROR::Device.py::log_retcode] RTLSDR-OPEN (#0): Error Code: -1 (-1)
[INFO::Device.py::stop_rx_mode] RTLSDR: Stopping RX Mode: Dialog closed. 
Killing recording process.

***@***.***:~$ sudo urh
[sudo] Passwort für dietmar:
Testing for explicit PulseAudio choice...
Testing for ALSA permissions...
...use PulseAudio this time and ask next time.
mkdir: cannot create directory '/run/user/0': Permission denied
QStandardPaths: XDG_RUNTIME_DIR points to non-existing path 
'/run/user/0/snap.urh', please create it with 0700 permissions.
No protocol specified
qt.qpa.screen: QXcbConnection: Could not connect to display :0
Could not connect to any X display.

andynoack, 2021-05-19:
Ok, maybe it's due to snap. Let's exclude that! Please try the git version or an installation with pip.

dwarning, 2021-05-20:
OK, deinstalled the snap version and made the pip install.

But now:

***@***.***:~$ urh
Traceback (most recent call last):
   File "/usr/local/bin/urh", line 8, in <module>
     sys.exit(main())
   File "/usr/local/lib/python3.8/dist-packages/urh/main.py", line 84, 
in main
     import urh.cythonext.signal_functions
   File "src/urh/cythonext/signal_functions.pyx", line 1, in init 
urh.cythonext.signal_functions
ValueError: numpy.ndarray size changed, may indicate binary 
incompatibility. Expected 88 from C header, got 80 from PyObject

I have a fresh Ubuntu 20.04 with all the actual ingredients like numpy, 
cython etc.

dwarning, 2021-05-20:
What a nightmare.

The problem was probably the numpy version 1.17 from the standard ubuntu 
distribution.

Found

pip3 install --upgrade numpy

to come to version 1.20.

Now URH runs and the dongle was found. Problem solved.

Thank you for your help.

andynoack, 2021-05-20:
Sometimes it's the little things...Nice that it works now, closing here!

closed 2021-05-20 by andynoack