jopohl/urh #881: usrp  X310 error

verss2, 2021-06-08:
The following error is displayed when I start URH.
[image]

Ubuntu 20.04 LTS 
UHD 3.15
URH  2.9.4
 Can you tell me how to solve this problem?
uhd_usrp_probe and  gnuraio3.8.2 work correctly .
URH running USRP b210 works fine.

andynoack, 2021-06-10:
We do not have an USRP X310 at our lab and I think Johannes has not access to one neither. In this case we need support from the community.

verss2, 2021-06-10:
> We do not have an USRP X310 at our lab and I think Johannes has not access to one neither. In this case we need support from the community.

Thank you for your reply. Have you ever tested the connection of X310? Can you tell me the system environment you are using? Also which files I modify to help add X310 support.

andynoack, 2021-06-10:
I have not had my hands on a X310. As far as I know (Johannes has cared for the UHD Driver), we we just use the UHD lib and we made our practical tests with the N210 series only.

jopohl, 2022-03-07:
So according to [the docs](https://files.ettus.com/manual/error_8h.html#a7fbc3f872dadd2c73cd656b2e5acdd2ba59829d0fc28459df62b18c88122a3d70) error code 40 refers to an assertion error. There has been some updates in UHD since this issue was reported, most recent version is [4.1.05](https://github.com/EttusResearch/uhd/releases/tag/v4.1.0.5) - there is a good chance this version fixes the reported error. Can someone from the community test the X310 in URH with a more recent UHD version?

jopohl, 2023-10-28:
I will close here due to inactivity. Feel free to reopen if the issue persists.

closed 2023-10-28 by jopohl