jopohl/urh #882: UHR 2.9.2 not working on Kali Linux 2021-01

maichai, 2021-06-10:
##### Expected Behavior

when starting uhr on Kali Linux 2021-01 it should open the main window (it all works under macOS for me).

##### Actual Behavior

```
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/urh/main.py", line 84, in main
    import urh.cythonext.signal_functions
  File "src/urh/cythonext/signal_functions.pyx", line 1, in init urh.cythonext.signal_functions
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
```

##### Steps To Reproduce

1. install uhr
2. start uhr

##### Screenshots

##### Platform Specifications

 - OS: Kali Linux 2021-01
 - URH version: 2.9.2
 - Python version: 3.9.2
 - Installed via pip3

Spessmen14, 2021-06-17:
I think i have the same issue

jopohl, 2021-06-17:
Seems to be the same issue as here (same fix should apply): https://github.com/jopohl/urh/discussions/878

If you want to prevent this from happening please convince the Kali guys to take URH into their repos. This is a classic packaging error.

closed 2021-06-17 by jopohl