jopohl/urh #886: Can't install from source

Spessmen14, 2021-06-17:
┌──(kali㉿debian)-[~/…/bin/urh/src/urh]
└─$ git clone https://github.com/jopohl/urh/
cd urh
python setup.py install
Cloning into 'urh'...
remote: Enumerating objects: 23811, done.
remote: Counting objects: 100% (47/47), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 23811 (delta 15), reused 12 (delta 4), pack-reused 23764
Receiving objects: 100% (23811/23811), 50.57 MiB | 5.04 MiB/s, done.
Resolving deltas: 100% (17178/17178), done.
  File "setup.py", line 44
    file=sys.stderr)

jopohl, 2021-06-17:
That output is a bit weird. What does 
```
python --version
```
give you? I would guess it is python2. If that is the case try
```
python3 setup.py install
```

NullVibes, 2021-06-24:
MacOs Catalina 10.15.7

I'm also unable to install from source (via this repo), or pip3 (which I'm guessing pulls from here, too.)  Brew install fine, but without support for RTL-SDR.  (Not selectable as source.)

Output shows error:
src/urh/dev/native/lib/sdrplay.cpp:4026:45:  error: no matching function for call to 'mir_sdr_StreamInit' ...