jopohl/urh #897: trouble installing urh on ubuntu 20.04

geo207, 2021-08-04:
Hello,

I followed the installation instructions on the URH site by installing pip and then using pip to install urh.  At the end of that process, the following line appeared in terminal:

Successfully installed cython-0.29.24 psutil-5.8.0 urh-2.9.2

I thought that means I installed it successfully.  However, when I entered urh in terminal in order to open the program, I received the following output:

$ urh
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/urh/main.py", line 84, in main
    import urh.cythonext.signal_functions
  File "src/urh/cythonext/signal_functions.pyx", line 1, in init urh.cythonext.signal_functions
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject

Was I right to enter urh in order to open the program, or should there be a different input?  If that was the right input, then what went wrong in the installation?  Thank you for your help.

jopohl, 2021-08-04:
Did you see #878? Should be the same issue, so the fix should be as simple as 
```
pip uninstall numpy
pip install numpy
```

geo207, 2021-08-04:
I didn't see it until now, but you're right it seems to be the same issue.  He was using Linux Mint 20.1, not Ubuntu.
The fix he recommends seems kind of drastic and I am getting the following caution:

$ pip uninstall numpy
Found existing installation: numpy 1.17.4
Uninstalling numpy-1.17.4:
  Would remove:
    /usr/bin/f2py
    /usr/bin/f2py3
    /usr/bin/f2py3.8
    /usr/lib/python3/dist-packages/numpy
    /usr/lib/python3/dist-packages/numpy-1.17.4.egg-info
Proceed (Y/n)?

Does this really seem like the right solution, and if I reinstall numpy with pip will I be putting back all of the programs that were removed?