jopohl/urh #900: Move bad signals to other group

hllhll, 2021-08-04:
##### Is your feature request related to a problem?
This ticket It's basically some way to circumvent my issue in https://github.com/jopohl/urh/issues/899 please see there

##### Describe the solution you'd like
An alternative way to handle the data that is acceptable, is only using the data that was properly decoded. The only way that makes sense for me is that I'd use the synchronization and preamble as a "guide" of what are good messages and what aren't
I'd like to be able to move them to a separate group, or better yet, delete them.

I.E. If I'm capturing hundreds of bursts, I'd like to filter/move according to rather "preamble was not found" or was found

[image]

##### Additional context
https://github.com/jopohl/urh/issues/899

andynoack, 2021-08-04:
You can already filter for the right preamble (everything else is temporarily not shown then), does that fix your problem?

hllhll, 2021-08-05:
Whoo, Missed that! ^_^

closed 2021-08-05 by hllhll