jopohl/urh #902: LimeSDR Mini fails to be opened as TX and RX during simulation

piotrva, 2021-08-12:
##### Expected Behavior
When starting a Simulation in URH using LimeSDR Mini as an TxRx URH should run it smoothly

##### Actual Behavior
When starting a Simulation in URH using LimeSDR Mini as an TxRx URH fails to do so, displaying error: 
Stop simulation (failed to start tx mode).
Using very same settings for receiving and transmitting separately works just fine.
Very same issue also happens under Windows 10 x64 - seems like URH tries to open FT601 twice (once for TX, once for RX), instead of providing single connection to the device.

##### Steps To Reproduce
1. Connect LimeSDR mini to PC
2. Start URH, open project
3. Prepare simple simulation
4. Start simulation

Information in Status window:
`Aug 12 02:13:38.497632: Waiting for devices
Aug 12 02:13:38.809148: Stop simulation (failed to start tx mode)`

Information in Sevices window:
`LimeSDR-OPEN: Error Code: -1 (-1)
LimeSDR: failed to start tx mode
LimeSDR-OPEN: Success
LimeSDR-INIT: Success
LimeSDR-SET_CHANNEL_INDEX to 0: Success
LimeSDR-SET_ANTENNA_INDEX to 3: Success
LimeSDR-SET_FREQUENCY to 2,476G: Success
LimeSDR-SET_SAMPLE_RATE to 8M: Success`

Log in the console:
`piotrva@PiotrMint:~$ urh
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
[INFO::settings.py::get_receive_buffer_size] Allocate receive buffer with 400MB
[INFO::Device.py::start_rx_mode] LimeSDR: Starting RX Mode
[INFO::Device.py::start_tx_mode] LimeSDR: Starting TX Mode
[DEBUG::Simulator.py::log_message] <i>Waiting for devices</i>
Error getting usb descriptor
Cannot claim interface - Resource busy
Failed to open device
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
Failed to open. Device is busy.
Unable to open device
Device cannot be NULL.
Device cannot be NULL.
Device cannot be NULL.
Device cannot be NULL.
[ERROR::Device.py::log_retcode] LimeSDR-OPEN: Error Code: -1 (-1)
[DEBUG::Simulator.py::log_message] Stop simulation (failed to start tx mode)
[INFO::Device.py::stop_rx_mode] LimeSDR: Stopping RX Mode: Stopping receiving due to user interaction
Reference clock 40,00 MHz
[INFO::Device.py::log_retcode] LimeSDR-OPEN: Success
Invalid channel number.
Invalid channel number.
[INFO::Device.py::log_retcode] LimeSDR-INIT: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_CHANNEL_INDEX to 0: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_ANTENNA_INDEX to 3: Success
[INFO::Device.py::log_retcode] LimeSDR-SET_FREQUENCY to 2,476G: Success
[INFO::Device.py::stop_rx_mode] LimeSDR: Stopping RX Mode: Stopping receiving due to user interaction
[INFO::Device.py::log_retcode] LimeSDR-SET_SAMPLE_RATE to 8M: Success
[WARNING::Device.py::stop_rx_mode] LimeSDR: Receive process is still alive, terminating it
[ERROR::Device.py::stop_rx_mode] [Errno 9] Bad file descriptor
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/urh/dev/native/Device.py", line 613, in stop_rx_mode
    connection.close()
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 177, in close
    self._close()
  File "/usr/lib/python3.8/multiprocessing/connection.py", line 361, in _close
    _close(self._handle)
OSError: [Errno 9] Bad file descriptor
[INFO::Device.py::read_device_messages] Exiting read device message thread due to [Errno 104] Connection reset by peer
[INFO::Device.py::stop_tx_mode] LimeSDR: Stopping TX Mode: EndlessSender stopped.
[INFO::Device.py::read_receiving_queue] EOF Error: Ending receive thread
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[INFO::Device.py::stop_tx_mode] LimeSDR: Stopping TX Mode: EndlessSender stopped.
[DEBUG::Device.py::read_receiving_queue] Exiting read_receive_queue thread.
[INFO::Device.py::read_device_messages] Exiting read device message thread due to [Errno 104] Connection reset by peer
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread`

##### Screenshots
Devices window with error:
[image]
TX settings:
[image]
RX settings:
[image]

##### Platform Specifications

 - OS: Linux Mint Linux PiotrMint 5.4.0-74-generic #83-Ubuntu SMP Sat May 8 02:35:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
 - URH version: 2.9.1
 - Python version: 3.8.10
 - Installed via pip
 Commands to install starting from clean OS:
`sudo apt install gnuradio gr-osmosdr

sudo apt install gnuradio gr-osmosdr limesuite liblimesuite-dev limesuite-udev limesuite-images cubicsdr soapysdr-module-lms7

LimeUtil --find

sudo python3 -m pip --upgrade pip
sudo python3 -m pip install urh
sudo python3 -m pip install --upgrade numpy

sudo apt install xterm
sudo apt install gr-limesdr
`

andynoack, 2021-08-12:
Most common devices do not have full duplex support, e.g. HackRF. Because of that we decided to make the general device backend of URH half duplex. That is why there will always be two connections (one for RX and for TX).

piotrva, 2021-08-12:
Are there any chances to get this usecase working? It would be natural to use full-duplex device in simulation mode ...

Maybe a workaround would be to allow for UDP send/receive, so it could be paired with UDP source/sink in gnuradio.

andynoack, 2021-08-12:
UDP source/sink maybe a solution for this use case, maybe also an adapted (i.e. replaced) GRC-Script for the communication with GnuRadio!

A complete rewrite of the URH backend to support full duplex in general (for the minority of SDRs) is currently not on our to do list. If you or someone else likes to implement that, we would be happy to accept the PR!

piotrva, 2021-08-12:
Thanks for the replay - I have been already looking at the code - both NetworkSDR plugin and LimeSDR driver - If I would have some time I would probably opt for doing UDP version of NetworkSDR first, as it would give really unlimited possibilities to pair URH with fully-custom GnuRadio interfaces (i.e. everything you can run in GnuRadio you can use with URH)