jopohl/urh #903: USRP: failed to start rx mode

liuyg851119ase, 2021-08-24:
Universal Radio Hacker (URH) choose USRP record radio signal,there was an error as followings：
[ERROR::Device.py::log_retcode] USRP-OPEN (type=b200,name=MyB210,serial=30FB59C,product=B210): Error Cod                                                                                   e: 44 (44)

USRP-OPEN (type=b200,name=MyB210,serial=30FB59C,product=B210): Error Code: 44 (44)
USRP: Could not retrieve more detailed error message from device.
USRP: failed to start rx mode

andynoack, 2021-08-24:
Please install from source and use the latest USRP drivers. This error comes directly from the driver, not from URH.

liuyg851119ase, 2021-08-24:
> Please install from source and use the latest USRP drivers. This error comes directly from the driver, not from URH.

Hi andynoack,
    I start from source code in command line(./main.py).
    I have install the uhd(the driver) and GNU radio and essential lib(I execute "sudo ldconfig" command ).
    But when I execute main.py in command line. There was an error as following:
    liuyg@vtk:~/decore/bus_key_radio/urh/src/urh$ ./main.py
Could not find C++ extensions, trying to build them.
Skipping native support for airspy
Skipping native support for bladerf
Skipping native support for hackrf
Skipping native support for limesdr
Skipping native support for plutosdr
Skipping native support for rtlsdr
Found uhd lib. Will compile with native usrp support
Skipping native support for sdrplay
Compiling src/urh/cythonext/signal_functions.pyx because it changed.
Compiling src/urh/cythonext/util.pyx because it changed.
Compiling src/urh/dev/native/lib/usrp.pyx because it changed.
[1/3] Cythonizing src/urh/cythonext/signal_functions.pyx
Traceback (most recent call last):
  File "setup.py", line 138, in <module>
    ext_modules=get_extensions(),
  File "setup.py", line 97, in get_extensions
    extensions = cythonize(extensions, compiler_directives=COMPILER_DIRECTIVES, compile_time_env=device_extras)
  File "/usr/lib/python3/dist-packages/Cython/Build/Dependencies.py", line 1027, in cythonize
    cythonize_one(*args)
  File "/usr/lib/python3/dist-packages/Cython/Build/Dependencies.py", line 1132, in cythonize_one
    result = compile([pyx_file], options)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Main.py", line 695, in compile
    return compile_multiple(source, options)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Main.py", line 673, in compile_multiple
    result = run_pipeline(source, options, context=context)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Main.py", line 494, in run_pipeline
    err, enddata = Pipeline.run_pipeline(pipeline, source)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Pipeline.py", line 341, in run_pipeline
    data = phase(data)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Pipeline.py", line 53, in generate_pyx_code_stage
    module_node.process_implementation(options, result)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/ModuleNode.py", line 141, in process_implementation
    self.generate_c_code(env, options, result)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/ModuleNode.py", line 377, in generate_c_code
    self.generate_module_init_func(modules[:-1], env, globalstate['init_module'])
  File "/usr/lib/python3/dist-packages/Cython/Compiler/ModuleNode.py", line 2234, in generate_module_init_func
    self.body.generate_execution_code(code)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Nodes.py", line 425, in generate_execution_code
    stat.generate_execution_code(code)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/Nodes.py", line 425, in generate_execution_code
    stat.generate_execution_code(code)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/FusedNode.py", line 831, in generate_execution_code
    default.generate_evaluation_code(code)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/ExprNodes.py", line 13041, in generate_evaluation_code
    self.arg.generate_evaluation_code(code)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/ExprNodes.py", line 11202, in generate_evaluation_code
    self.generate_div_warning_code(code)
  File "/usr/lib/python3/dist-packages/Cython/Compiler/ExprNodes.py", line 11205, in generate_div_warning_code
    in_nogil = self.in_nogil_context
AttributeError: 'DivNode' object has no attribute 'in_nogil_context'
Traceback (most recent call last):
  File "./main.py", line 189, in <module>
    main()
  File "./main.py", line 100, in main
    from urh.controller.MainController import MainController
  File "/home/liuyg/decore/bus_key_radio/urh/src/urh/controller/MainController.py", line 9, in <module>
    from urh.controller.CompareFrameController import CompareFrameController
  File "/home/liuyg/decore/bus_key_radio/urh/src/urh/controller/CompareFrameController.py", line 14, in <module>
    from urh.awre import AutoAssigner
  File "/home/liuyg/decore/bus_key_radio/urh/src/urh/awre/AutoAssigner.py", line 3, in <module>
    from urh.cythonext import util
ImportError: cannot import name 'util'
   But there are util.pxd and util.pyx in urh/cythonext directory. Is it the path issue?

andynoack, 2021-08-24:
It rather looks like wrong or missing dependencies for me, as util.pyx seems not to compile ordinary.

liuyg851119ase, 2021-08-24:
> It rather looks like wrong or missing dependencies for me, as util.pyx seems not to compile ordinary.

   "Could not find C++ extensions, trying to build them." How can I install C++ extensions.

andynoack, 2021-08-24:
This depends on your distribution, on Ubuntu it should work with apt or pip. Another possibility is using the URH docker container. I will close here because this is not directly related to URH anymore. You can ask on our slack channel for installation support.

closed 2021-08-24 by andynoack