jopohl/urh #904: Airspy mini sample rates

dwarning, 2021-08-24:
##### Expected Behavior
Airspy mini sample rate is specified  with 10, 6 and 3 Msample/s.
Expected behavior would be that I can choose one of these in URH.

##### Actual Behavior
Only 6 Msample/s is usable URH.

##### Platform Specifications
 - OS: Win 10
 - URH version: 2.9.2
 - Python version: 3.7
 - Installed via [e.g. pip]

andynoack, 2021-08-25:
There seems to be a bug in the driver, so we only allow 6MS/s to prevent misunderstandings or logical bugs.

See config.py:114 "# Documentation says: "10, 6 and 3 MSPS IQ output" but it always uses 6M, no matter what is configured."

closed 2021-08-25 by andynoack

dwarning, 2021-08-28:
I made this trial.
First I checked with airspy_info the provided sample rates: 3M and 6M.
Then I changed config.py:
`    "center_freq": dev_range(start=24 * M, stop=1800 * M, step=1),
    "sample_rate": [3 * M, 6 * M],
    # Documentation says: "10, 6 and 3 MSPS IQ output" but it always uses 6M, no matter what is configured.
    "bandwidth": [3 * M, 6 * M],
`
and Airspy.py line 51
`        self.bandwidth_is_adjustable = True`
I checked the gathered samples, OK. But a bit unsure regarding bandwidth.

One open question:
I think there is a misunderstanding with the Airspy provided centre frequencies:
24 ... 1800MHz in the spec means 24MHz as low end and not 24Hz as in the actual git code.
Please comment!
Thank you

andynoack, 2021-08-28:
Your changes seem sound, I think you can do it like this! You may even be right about 24Hz vs. 24MHz, but it is still not 100% clear from the website.
I wouldn't change that in URH/master without thorough research. There is also the problem that with at least some library versions there are problems with other sample rates than 6M.
If the changes work with your setup, that is great!

Prof. Dr.-Ing. Andreas Noack
Hochschule Stralsund

dwarning, 2021-08-28:
Agree with further investigations.

But with 24Hz lower end it would not make sense that they sell their 
up-converter named "SpyVerter" to cover low frequency range.

Kind Regards,

Dietmar Warning