jopohl/urh #905: HackRF not transmitting

careyer, 2021-08-28:
##### Expected Behavior
Recording a signal with URH and HackRF seems to work fine (I can clearly see the signal in the analog signal view).
However when replaying the exact same signal via URH there is no reaction of the target (garage door) at all. 

##### Actual Behavior
It actually seems like the HackRF does not transmit anything via URH. I have tried to record and playback the signal at various sample rates and bandwidth but non of these showed any effect unpon replay. The issue seems similiar to what has been reported in #664 

##### Steps To Reproduce

1. Record a signal (in my case 27,195Mhz, 2M sample rate, 2M bandwidth)
2. See that the analog signal view destinctivly shows the signal
3. Go ahead and replay the signal 
4. Notice that  upon replay the target (in my case garage door) will stay unresponsive
5. **_Crosscheck with HackRF Portapack Mayhem that indeed recording and playback of the signal works just fine (see sidenotes)_**

##### Screenshots

[image]
Recorded signal destinctively shows in the analog signal view

##### Platform Specifications

 - OS: Windows 10 65bit
 - URH version: 2.9.2
 - Python version: 3.8.10

#### Sidenotes:
 -  Capturing the very same signal with my HackRF Portapack (Mayhem) and replaying the signal from the portapack the garage door opens and closes as expected without any problem. This shows that a hardware defect and other obstacles (like e.g. rolling code) can be ruled out.  
 - I noticed that the sample rate for the HackRF cannot be lowered lower than 2M for the HackRF with URH. With HackRF Portapack I captured the signal with only 25k samples/sec which is a considerably lower sample rate but obviously suffcient to gain the expected success (would be nice to lower this limit in URH). 
 - I also tried to load the RF-data captured with HackRF Portapack into URH but sadly this does not work. It somehow does missinterpret the file format. I know that this is a feature request and I may open one up seperatly here but it would be handy if URH could import the HackRF Portapack recordings as is. Since the portapack (add-on board to the HackRF) is widely spread now and readily available for cheap  this is the go to solution to capture signals without the need of a connected PC/Laptop in the field. It would be awesome to have support for this. I will include a sample of the file format here for reference: 
[Garage_IN.zip](https://github.com/jopohl/urh/files/7071285/Garage_IN.zip)

Other than that.. please keep up you awesome work on this software marvel!
Kudos! And all the best
Cheers!
Thomas

andynoack, 2021-08-28:
Hey Thomas,
we made similar experiences with the HackRF. In our case it was just the gain that was not sufficient. Replay is pretty straight forward, no Interpretation in URH (an error is rather unlikely here). Additionally your selected frequency is pretty low, is your antenna suited for TX in this band?

careyer, 2021-08-29:
> Hey Thomas,
> we made similar experiences with the HackRF. In our case it was just the gain that was not sufficient. Replay is pretty straight forward, no Interpretation in URH (an error is rather [...]

Hi @andynoack. Thank you for stepping in an trying to help on this issue. 
I indeed tried to crank up the gains (tried both Gain and IF-Gain) but to no avail. 

Yes frequency is low... it is 27.195Mhz but my antenna is suiteable. With the Portapack I was able to replay the signal successfully and the range in which the garage door would open was about 10 meters, so the signal strength can't be too shabby with that antenna.

**A thing that I noticed though**: With URH I can only set the Gain to either  the value "1" or "14", whereas on the Portapack this value has a range of 0-40. Why is that? Maybe that is where it breaks? On Portapack during replay I can select the LNA gain (0-40) and I can tick a checkbox that sais Amplifier (on/off). Garage door opens just fine at LNA=32 Amplifier=off.

andynoack, 2021-08-29:
I think this is matter of software interfaces. When we implemented this URH component we looked up, what the hardware in cooperation with the current firmware and software library is able to do (e.g. gain 1 and 14). Other programs provide other settings, maybe with percentages, but we wanted to be as honest as possible. When you are in direct control of the firmware (with portapack), however, you may have options that a SDR capable software has not.
For similar problems it was always the gain for us. Is it possible that you try the replay attack near the garage door (using different gain settings) with URH?

careyer, 2021-08-29:
I also tried  to perform the replay in direct proximity… to the receiver. Literally only half a meter away but sadly nothing. It is as if it is not transmitting at all. I am sorry to not have any better news :-/

Maybe only specific firmware versions for the HackRF are supported and mine is incompatible? Only updated a few weeks back

Cheers!

andynoack, 2021-08-29:
That could be the case. Have you tried to recompile the Cython-Extensions (cythonext/build.py)?
Another possibility would be try linux instead of Windows.

careyer, 2021-08-29:
Thank you very much for the suggestions.. I wont have time the comming days to test this again since I am on vactation but I suggest to leave this issue open for now to see if other people have similar issues, are you okay with that? I also own a Ettus B210 board with which I can confirm if the HackRF indeed is transmitting anything or not. However the frequency tange of the B210 starts only at 70Mhz whereas the HackRF starts at 1Mhz. ... That is why I could not test this yesterday with the 27.195Mhz.