jopohl/urh #908: ImportError: cannot import name 'usrp' from 'urh.dev.native.lib'

gitbobo, 2021-09-28:
##### Expected Behavior
running urh correctly

##### Actual Behavior
```
zi@new:$ python3 /home/zi/.local/lib/python3.7/site-packages/urh/cli/urh_cli.py -d USRP -f 100e6 -s 40e6 -rx -mo FSK -pm 0
Traceback (most recent call last):
  File "/home/zi/.local/lib/python3.7/site-packages/urh/cli/urh_cli.py", line 516, in <module>
    main()
  File "/home/zi/.local/lib/python3.7/site-packages/urh/cli/urh_cli.py", line 472, in main
    receiver = build_protocol_sniffer_from_args(args)
  File "/home/zi/.local/lib/python3.7/site-packages/urh/cli/urh_cli.py", line 148, in build_protocol_sniffer_from_args
    arguments.device.lower(), bh)
  File "/home/zi/.local/lib/python3.7/site-packages/urh/signalprocessing/ProtocolSniffer.py", line 48, in __init__
    resume_on_full_receive_buffer=True, raw_mode=network_raw_mode)
  File "/home/zi/.local/lib/python3.7/site-packages/urh/dev/VirtualDevice.py", line 120, in __init__
    from urh.dev.native.USRP import USRP
  File "/home/zi/.local/lib/python3.7/site-packages/urh/dev/native/USRP.py", line 8, in <module>
    from urh.dev.native.lib import usrp
ImportError: cannot import name 'usrp' from 'urh.dev.native.lib' (/home/zi/.local/lib/python3.7/site-packages/urh/dev/native/lib/__init__.py)  
```

##### Steps To Reproduce

1. zi@new:$ python3 -m pip install urh
2. zi@new:$ python3 /home/zi/.local/lib/python3.7/site-packages/urh/cli/urh_cli.py -d USRP -f 100e6 -s 40e6 -rx -mo FSK -pm 0

##### Screenshots

```
zi@new:$ python3 -m pip install urh
Collecting urh
  Using cached https://pypi.repo.net/packages/06/d8/f140e9c0f592134580819b959121b47bce042694168032bf8b219d39c977/urh-2.9.2.tar.gz
Requirement already satisfied: cython in /usr/lib/python3/dist-packages (from urh) (0.29.2)
Collecting numpy (from urh)
  Downloading https://pypi.repo.net/packages/8f/3d/7247b219c2c3bc5720a0ae963bf384df1f7f39c17330fcff2bd9d8640768/numpy-1.21.2-cp37-cp37m-manylinux_2_5_x86_64.manylinux1_x86_64.whl (14.1MB)
    100% |████████████████████████████████| 14.1MB 70kB/s 
Collecting psutil (from urh)
  Downloading https://pypi.repo.net/packages/e1/b0/7276de53321c12981717490516b7e612364f2cb372ee8901bd4a66a000d7/psutil-5.8.0.tar.gz (470kB)
    100% |████████████████████████████████| 471kB 349kB/s 
Building wheels for collected packages: urh, psutil
  Running setup.py bdist_wheel for urh ... done
  Stored in directory: /home/zi/.cache/pip/wheels/44/41/eb/00f969004d3ac4b7e1b1ea151a075e2587c6dbf29c5d22b222
  Running setup.py bdist_wheel for psutil ... done
  Stored in directory: /home/zi/.cache/pip/wheels/08/fc/46/4784535f463b8d4d923d06e96816a609a49286f3ee742b9d03
Successfully built urh psutil
Installing collected packages: numpy, psutil, urh
  The scripts f2py, f2py3 and f2py3.7 are installed in '/home/zi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The scripts urh and urh_cli are installed in '/home/zi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed numpy-1.21.2 psutil-5.8.0 urh-2.9.2
``` 

##### Platform Specifications

 - OS: [debian 10]
 - URH version: [2.9.2]
 - Python version: [e.g. 3.7]
 - Installed via [pip]

andynoack, 2021-09-29:
I just tested it over here but sadly cannot reproduce your error. Have you tried using a recent python3 version?

gitbobo, 2021-09-29:
> I just tested it over here but sadly cannot reproduce your error. Have you tried using a recent python3 version?

Default python is 3.7 in debian 10,I will try 3.8 later and feedback to you.