jopohl/urh #914: Not working since portapack mayhem

hexan86, 2021-11-08:
##### Expected Behavior
Load the device

##### Actual Behavior
Crashes

##### Steps To Reproduce

1. Go to 'Spectrum Analyzer' or Other
2. Click on 'Device' and select HackRF
3. Click on the Green arrow on the side (Auto detect)
4. URH crashes

##### Platform Specifications
 - OS: Windows 10
 - URH version: 2.9.2
 - Python version: 3.9

Since when I've added the PortaPack H2+ (Mayhem Firmware) to my HackRF I'm struggling a bit. SDR Sharp is not working anymore and now URH too. Other softwares are working just fine, but URH is not able to work anymore.
I'm obviously selecting the "HackRF" mode on the PortaPack display, connecting it to my Win 10 Machine, but then, if I try to use it, there is nothing to do.

jopohl, 2022-02-22:
URH uses the default hackrf lib under the hood. I do not have a HackRF anymore and never owned a PortaPack. So we would need some help from the community to get this working.

mrweaver, 2022-03-04:
Hey guys,
I am experiencing a very similar issue with the HackRF, currently running Mayhem 1.4.3 on the PortaPack. I've done a screen grab, but basically it will run for a couple of seconds, before hanging and I am forced to crash URH.

Have you had any luck, @hexan86?

**Platform Specifications**
OS: Windows 10
URH version: 2.9.3
Python version: 3.10.2

Cheers,
Michael.

[image]

hexan86, 2022-03-04:
I have removed Zadig drivers and now it works just fine.

Now I can use it as portapack (obviously) and connected to my pc, with URH and other softwares.

mrweaver, 2022-03-04:
Thanks for getting back to me, I'm glad to hear that you got it working.

I've removed and deleted the drivers, and installed a few variants without
any change in behaviour.
 Are you able to tell me which driver you currently have installed so that
I can try that?

hexan86, 2022-03-04:
@mrweaver I have now the Sharebrained driver, dated 08/10/2015, v.1.0.0.0 

Driver details:
system32\
-DRIVERS\winusb.sys
-WdfCoInstaller01011.dll
-WinUSBCoInstaller2.dll

In "Device management", I don't see it under "computer->USB Controller", but "Computer->USB Devices->HackRF One"

mrweaver, 2022-03-04:
@hexan86, thanks for the info there.
I installed that same driver, I see it's the one that came with the Mayhem firmware. Unfortunately, I am still having the same issue.

There is something else going on here. What else could be affecting this?