jopohl/urh #916: LimeSDR - Simulator - duplicate messages

prefixFelix, 2021-11-19:
#### Expected Behavior
In simulation mode, several messages are to be transmitted. In this example, a delay of 5 seconds is set before the message frame to show the problem more clearly (it also occurs without a delay).

##### Simulator view:
[image]

#### Actual Behavior
If you start the simulation for the first time, everything runs as expected:
##### First start of the simulation inspected with Gqrx:
[image]

##### First start of the simulation recorded with URH:
[image]

If you press the start button again in the simulation window, you will get this result:
##### Second start of the simulation inspected with Gqrx:
[image]

##### Second start of the simulation recorded with URH:
[image]

Parts the message frame (or the whole frame) are sent both shortly after the calibration phase of the LimeSDR as well as at the correct time slot. The time delay is irrelevant. If you simply remove it, more messages will be sent than are specified in the flow diagram. If you press the start button again, you will get the same result.  
Only by opening and closing the simulation window is it possible to send an valid message block again.

#### Steps To Reproduce

1. Use LimeSDR
2. Create a simple flow diagram as shown above
2. Run the simulations 2 times in a row

#### Platform Specifications

 - OS: Ubuntu 21.04 / EndeavourOS 2021.08.27
 - URH version:  2.9.2 / 2.9.2
 - Python version: 3.9.5 / 3.9.7
 - Build form source / snap

andynoack, 2021-11-20:
I guess this is due to not flushing the TX buffer (fast enough) after sending...I am not even sure whether we could possibly fix that in URH.

prefixFelix, 2021-11-20:
Ok, I guess it's not that big a bug, since you can just reopen the window to fix it.

andynoack, 2021-11-20:
You are welcome to make a PR, if you think, that one is easy to do. At the moment I don't have my LimeSDR ready, so we would really appreciate any help here.