jopohl/urh #917: HackRF - Simulator not working

prefixFelix, 2021-11-19:
#### Expected Behavior
If I send the same messages with exactly the same settings once in the generation mode and once in the simulation mode, the result should be the same. 

##### Settings:
[image]

#### Actual Behavior
In the generator, the messages are transmitted as expected, but in the simulator, no/wrong messages are transmitted. If you look closer at the USB traffic between URH and HackRF you will notice that in simulator mode almost exclusively 0x00 IQ samples are transmitted. Sometimes you can find edge values like 0xff or 0x01.

##### Generator IQ-Samples as 8 bit signed int:
[image]

##### Generation inspected with Gqrx:
[image]

##### Simulator IQ-Samples as 8 bit signed int:
[image]

##### Simulation inspected with Gqrx:
[image]

#### Steps To Reproduce

1. Use HackRF
2. Create a simple flow diagram
3. Run the simulation

```
# Generator
[DEBUG::GeneratorTabController.py::prepare_modulation_buffer] Allocating 3.20MB for modulated samples
[INFO::Device.py::start_tx_mode] HackRF: Starting TX Mode
[INFO::Device.py::log_retcode] HackRF-SETUP (0000000000000000466c64c8275347c7): Success
[INFO::Device.py::log_retcode] HackRF-SET_FREQUENCY to 433,92M: Success
[INFO::Device.py::log_retcode] HackRF-SET_SAMPLE_RATE to 2M: Success
[INFO::Device.py::log_retcode] HackRF-SET_BANDWIDTH to 2M: Success
[INFO::Device.py::log_retcode] HackRF-SET_RF_GAIN to 14: Success
[INFO::Device.py::log_retcode] HackRF-SET_IF_GAIN to 47: Success
[INFO::Device.py::log_retcode] HackRF-SET_BIAS_TEE_ENABLED to 0: Success
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: Stopped receiving: Stop button clicked
[DEBUG::Device.py::device_send] type: exit requested. Stopping sending
[DEBUG::Device.py::device_send] type: sending is finished.
[INFO::Device.py::log_retcode] HackRF-STOP TX MODE: Success
[INFO::Device.py::log_retcode] HackRF-CLOSE: Success
[INFO::Device.py::log_retcode] HackRF-EXIT: Success
[INFO::Device.py::read_device_messages] Exiting read device message thread due to 
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[DEBUG::Device.py::stop_tx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: Dialog closed. Killing recording process.
[DEBUG::SendRecvDialog.py::closeEvent] Device stopped successfully.
[DEBUG::SendRecvDialog.py::closeEvent] Cleaning up device
[DEBUG::SendRecvDialog.py::closeEvent] Successfully cleaned up device

# Simulator
[INFO::Device.py::start_tx_mode] HackRF: Starting TX Mode
[DEBUG::Simulator.py::log_message] <i>Waiting for devices</i>
[INFO::Device.py::log_retcode] HackRF-SETUP (0000000000000000466c64c8275347c7): Success
[INFO::Device.py::log_retcode] HackRF-SET_FREQUENCY to 433,92M: Success
[INFO::Device.py::log_retcode] HackRF-SET_SAMPLE_RATE to 2M: Success
[INFO::Device.py::log_retcode] HackRF-SET_BANDWIDTH to 2M: Success
[DEBUG::Simulator.py::log_message] <i>Waiting for devices</i>
[INFO::Device.py::log_retcode] HackRF-SET_RF_GAIN to 14: Success
[INFO::Device.py::log_retcode] HackRF-SET_IF_GAIN to 47: Success
[INFO::Device.py::log_retcode] HackRF-SET_BIAS_TEE_ENABLED to 0: Success
[DEBUG::Simulator.py::log_message] TX is ready to operate
[DEBUG::Simulator.py::log_message] <b>Simulation is running</b>
[DEBUG::Simulator.py::log_message] Sending message 1
[DEBUG::Simulator.py::log_message] Sending message 2
[DEBUG::Simulator.py::log_message] Sending message 3
[DEBUG::Simulator.py::log_message] Sending message 4
[DEBUG::Simulator.py::log_message] Stop simulation (Finished)
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: EndlessSender stopped.
[DEBUG::Device.py::device_send] type: exit requested. Stopping sending
[INFO::Device.py::log_retcode] HackRF-STOP TX MODE: Success
[INFO::Device.py::log_retcode] HackRF-CLOSE: Success
[INFO::Device.py::log_retcode] HackRF-EXIT: Success
[INFO::Device.py::read_device_messages] Exiting read device message thread due to 
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
[DEBUG::Device.py::stop_tx_mode] Closing parent control connection: handle is closed
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: EndlessSender stopped.
```

#### Platform Specifications
I tried it with two different HackRFs and also mixed the platforms listed below.

 - OS: Ubuntu 21.04 / Windows 10 / EndeavourOS 2021.08.27
 - URH version: 2.9.2, 2.0.2
 - Python version: 3.9.5 / 3.9.5 /  3.9.7 
 - Build form source / binary /  snap
 - HackRF firmware: 2021.03.1, 2018.01.1

andynoack, 2021-11-20:
The TX results from generator (except very long frames) and simulator differ due to technical reasons. The simulator needs to be as fast as possible whereas the generator does not care for delay. Consequently the simulator permanently sends data from a buffer which returns zero if empty.

prefixFelix, 2021-11-20:
Ah ok that makes sense. Would you still have an idea why no messages are transmitted?

andynoack, 2021-11-20:
Maybe have a look in the modulation settings, check the flow graph incl. decodings and the delay settings.

prefixFelix, 2021-11-22:
I have double checked everything, but the HackRF still does not want to send messages. I have also tried it with other flow graphs, frequencies and messages. I only use NRZ for the decoding.
Everything works as expected when I start the simulation with the LimeSDR. Might there be any dependencies that are causing the problem?

andynoack, 2021-11-22:
Well, most of our lab tests were done using HackRFs and we did not notice any problems with them, in contrast to some LimeSDRs that we have tested. Your report suggests that there is something wrong with the HackRF API. Have you performed a firmware upgrade or something like that recently? (Another issue brought up that there might be problems with custom firmwares.)
Have you already rebuilt the cython extensions (cythonext/build.py)? If this doesn't fix the problem, maybe @jopohl can have a look.

jopohl, 2021-11-22:
There is also the setting "Options" -> "Generation" -> "Modulation Accuracy" that might be worth experimenting with. Shouldn't make a difference, but will be good if we can rule this out.

prefixFelix, 2021-11-22:
> Have you performed a firmware upgrade or something like that recently? (Another issue brought up that there might be problems with custom firmwares.)

Yes I have upgraded the firmware of the HackRF to version 2021.03.1 recently (The driver is build form source from the last commit, but I also tried the releases 2021.03.1 and 2018.01.1).

> Have you already rebuilt the cython extensions (cythonext/build.py)? If this doesn't fix the problem, maybe @jopohl can have a look.

Unfortunately that did not help, but here is the log:
```
Skipping native support for airspy
Skipping native support for bladerf
Found hackrf lib. Will compile with native hackrf support
Found LimeSuite lib. Will compile with native limesdr support
Skipping native support for plutosdr
Skipping native support for rtlsdr
Skipping native support for usrp
Skipping native support for sdrplay
running build_ext
Finalizing options
copying build/lib.linux-x86_64-3.9/urh/cythonext/awre_util.cpython-39-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.9/urh/cythonext/util.cpython-39-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.9/urh/cythonext/signal_functions.cpython-39-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.9/urh/cythonext/path_creator.cpython-39-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.9/urh/cythonext/auto_interpretation.cpython-39-x86_64-linux-gnu.so -> src/urh/cythonext
copying build/lib.linux-x86_64-3.9/urh/dev/native/lib/hackrf.cpython-39-x86_64-linux-gnu.so -> src/urh/dev/native/lib
copying build/lib.linux-x86_64-3.9/urh/dev/native/lib/limesdr.cpython-39-x86_64-linux-gnu.so -> src/urh/dev/native/lib
```
>- The Generator works as expected with the HackRF
>-  The Simulator works in general but not with the HackRF specifically

Yes exactly.

> I am curios what is your use case for using the Simulator for TX only? As "RX settings" are grayed out in your first screenshot, I assume there is not RX involved?

In my usual project I use TX & RX as intended, but for troubleshooting it was more convenient for me to only use TX.

> There is also the setting "Options" -> "Generation" -> "Modulation Accuracy" that might be worth experimenting with. Shouldn't make a difference, but will be good if we can rule this out.

I think that did the trick! Wireshark shows that IQ-sampels are transmitted! The setting was set to high and in low mode something was transmitted. But I still do not understand exactly what the setting does, could you explain this further? I will check tomorrow with a second SDR whether really all messages are transmitted. 

Thank you both for the fast replies and for the great software!

jopohl, 2021-11-22:
> The setting was set to high and in low mode something was transmitted. But I still do not understand exactly what the setting does, could you explain this further?

SDRs differ in sampling accuracy. HackRF has the lowest one (8-bit integer for IQ, respectively), so if you would try to send a stream of 32-bit float IQ data to a HackRF it would explain the large amount of 0s and 1s as in terms of float IQ is modeled in intervals [-1, 1]. We have a somewhat related wiki page for that here: https://github.com/jopohl/urh/wiki/Supported-signal-file-formats

Having said that, if changing the setting repairs the Simulator this is a bug that should be fixed on URH side.

prefixFelix, 2021-11-23:
All right, so I can confirm that the simulator runs smoothly with the low setting:
[image]

jopohl, 2021-11-26:
Thanks for confirming, I created a fix in #918 - could you please checkout that branch and verify the simulation is working regardless of the "Modulation Accuracy" setting.

prefixFelix, 2021-11-30:
Yes it works now regardless of the setting. Thank you for the fix!

jopohl, 2021-11-30:
Cheers - merged that PR into master. Thanks for verification of the fix!

PR #918 (2021-11-30): Simulation fix: use sending device dtype for modulation
fixes #917
closed 2021-11-30 by jopohl