jopohl/urh #919: Simulator - Trigger Command - item.rc

prefixFelix, 2021-12-08:
##### Expected Behavior
The output of script _stdout.py_ should be accessible via the item identifier _item1.rc_.

##### Actual Behavior
URH detects the correct output (see log), but it cannot be accessed via item1.rc. Only eight 0's are set (which corresponds to the number of the actual field length). If you directly use the option "External Program" everything works as expected. 
I have also checked that it is not a display error.

###### Simulator flow graph:
[image]

###### Log:
```
[INFO::Device.py::start_tx_mode] HackRF: Starting TX Mode
[DEBUG::Simulator.py::log_message] <i>Waiting for devices</i>
[INFO::Device.py::log_retcode] HackRF-SETUP (0000000000000000466c64c8275347c7): Success
[INFO::Device.py::log_retcode] HackRF-SET_FREQUENCY to 433.92M: Success
[INFO::Device.py::log_retcode] HackRF-SET_SAMPLE_RATE to 2M: Success
[INFO::Device.py::log_retcode] HackRF-SET_BANDWIDTH to 2M: Success
[DEBUG::Simulator.py::log_message] <i>Waiting for devices</i>
[INFO::Device.py::log_retcode] HackRF-SET_RF_GAIN to 0: Success
[INFO::Device.py::log_retcode] HackRF-SET_IF_GAIN to 23: Success
[INFO::Device.py::log_retcode] HackRF-SET_BIAS_TEE_ENABLED to 0: Success
[DEBUG::Simulator.py::log_message] TX is ready to operate
[DEBUG::Simulator.py::log_message] <b>Simulation is running</b>
[DEBUG::Simulator.py::log_message] Calling stdout.py
[DEBUG::Simulator.py::log_message] /home/felix/Documents/URH-test/433Mhz/stdout.py exited with 0 stdout: 10101010
[DEBUG::Simulator.py::log_message] Sending message 2
[DEBUG::Simulator.py::log_message] Stop simulation (Finished)
[INFO::Device.py::stop_tx_mode] HackRF: Stopping TX Mode: EndlessSender stopped.
[DEBUG::Device.py::device_send] type: exit requested. Stopping sending
[INFO::Device.py::log_retcode] HackRF-STOP TX MODE: Success
[INFO::Device.py::log_retcode] HackRF-CLOSE: Success
[INFO::Device.py::log_retcode] HackRF-EXIT: Success
[INFO::Device.py::read_device_messages] Exiting read device message thread due to 
[DEBUG::Device.py::read_device_messages] Exiting read device errors thread
```
###### Simulator tab:
[image]

##### Steps To Reproduce
1. Write a simple script:
```
#!/usr/bin/env python3
import sys
sys.stdout.write("10101010")
```
2. Build a flow graph as shown above

##### Platform Specifications
 - OS: Ubuntu 21.04
 - URH version: 2.9.2
 - Python version: 3.9.5
 - Installed via snap / build from source

andynoack, 2021-12-08:
Maybe this is a misunderstanding, "rc" should be the return code and not the output of a script.

prefixFelix, 2021-12-08:
Ah ok now I understand. Maybe I should have noticed that earlier. So .rc is simply an integer which represents the exit code of the script? Is there a way to also interact with the output or is the "Trigger Command" function actually not meant for that?

andynoack, 2021-12-08:
Yes, exactly. "Trigger Command" is actually not meant to import data into URH, please use "External Program"s for this purpose.

prefixFelix, 2021-12-09:
All right thanks for the infos. I would like to submit a small addition to the Wiki:
One could mention that when using the "External Program" function, not the entire transcript is passed, but only the direction of the messages without the names of the participants (<- / ->).

andynoack, 2021-12-09:
Alright! I have modified the wiki accordingly. Thanks for the suggestion!

closed 2021-12-10 by andynoack