jopohl/urh #920: Add ShiftBits in Additional Functions of Decoding

KimIV, 2021-12-10:
I am researching TPMS sensor signals. And I found that before applying the Edge Trigger, it is necessary to shift all the bits to the right by one bit. Otherwise, Manchester decoding did not work correctly.

I see the solution in allowing the user to decide for himself which bit to start using Manchester decoding. Universal Radio Hacker is often mistaken in defining this bit. As a result, the decoder tries to decode such forbidden combinations as "00" and "11".

The ideal would be to choose yourself on the Interpretation tab from which time position to start defining the bits. Or be able to add leading zeros. Also, a solution could be to add a left / right bit shift function to the additional functions of the decoder.

I copied the original data like:
aaaaaaaccacd2ad2d2b4ab2aaaaaccb2ab332cb2
in Excel. Shifted them one bit to the right and got:
5555555665669569695a55955555665955999659
And then I used Manchester decode:
00014586630800520a92
And this data is already correct. In them:
45866308 - sensor ID,
00 - pressure,
52 - temperature
92 - checksum.

andynoack, 2021-12-10:
Why don't you use "Cut before/after" with position 1?

KimIV, 2021-12-10:
"Cut" doesn't work. For correct decoding, the code must start at 0 for 010101 and so on. But in reality, the code starts with 1, that is, 101010 ... And it would be correct to add the leading 0. Well, or shift the entire code one bit to the right.

And by the way, where is "Cut"? I did not find.

KimIV, 2021-12-10:
If I myself add a leading 0 in the code field, then it is not taken into account during decoding
[image]
[image]
[image]
[image]
.

andynoack, 2021-12-10:
Well, it does not work like this. Please craft a custom Manchester Decoding where you prepend the mentioned cut operation. As this is not an URH issue (Feature/Error) but a usage problem, please use our slack channel for further individual support.

closed 2021-12-10 by andynoack

Crsarmv7l, 2022-12-11:
> Well, it does not work like this. Please craft a custom Manchester Decoding where you prepend the mentioned cut operation. As this is not an URH issue (Feature/Error) but a usage problem, please use our slack channel for further individual support.

Lol. I just spent quite a bit of time with this issue. Love how "its user error" is the conclusion