jopohl/urh #921: Trojan in the urh.exe

graiSoft1982, 2021-12-12:
[image]

graiSoft1982, 2021-12-12:
https://www.virustotal.com/gui/file/8351dddb42dbb15511729bd01025fb8b71aba97b0dbf028b9141fecfe2e4f42a

Cloud-Forge, 2021-12-12:
[image]

Just uploaded the file to VirusTotal. It came out totally clean... It's surely a false positive on your side

Cloud-Forge, 2021-12-12:
> https://www.virustotal.com/gui/file/8351dddb42dbb15511729bd01025fb8b71aba97b0dbf028b9141fecfe2e4f42a

**is-RAIGO.tmp** 

What is this ? It's not from this repo ... 

This has nothing to do with URH.

graiSoft1982, 2021-12-12:
the virus is in the 32-bit exe file

Thomas ***@***.***> ezt írta (időpont: 2021. dec. 12., V
17:34):

andynoack, 2021-12-13:
URH is a pretty large project with a lot of low level and communication stuff. When compiled, I can imagine that some AV scanners find false positives.
I recommend to use the non compiled python version if you are irritated by the scanner result. By the way, the artifacts (urh.exe, ...) are compiled by GitHub and not uploaded by us. I will close this issue to prevent rumors.

closed 2021-12-13 by andynoack

jopohl, 2021-12-13:
To be more precise, we use PyInstaller/InnoSetup to produce the binaries. If you are curious how that works you can see our CI definition [here](https://github.com/jopohl/urh/blob/master/data/azure-pipelines.yml#L159-L164).
I answered on that matter a while ago in [this](https://github.com/jopohl/urh/issues/804#issuecomment-702195611) issue:

>This seems to be an issue with PyInstaller, let me quote the author here:
>> Please contact you anti-virus vendor. There is *nothing* we can do about this false positive.

>> If your anti-virus vendor considers one of the files included in the PyInstaller distribution or a file generated by PyInstaller to be malicious, there is *nothing* we can do about this. Even if we'd change our code, they'd change their pattern and the race starts again.

>> See [this mailing-list thread](http://www.mail-archive.com/pyinstaller@googlegroups.com/msg05002.html) and [other tickets for his topic](https://github.com/pyinstaller/pyinstaller/issues?q=is%3Aissue+virus+is%3Aclosed).

>_Originally posted by @htgoebel in https://github.com/pyinstaller/pyinstaller/issues/4633#issuecomment-688886522_

As @andynoack pointed out, you can install URH with pip or run it from source if you are concerned about this. We have nothing to hide.

htgoebel, 2021-12-13:
Long-time Pyinstaller maintainer here. My statement is still true: Please contact you anti-virus vendor. There is nothing PyInstaller or URH can do about this false positive.