jopohl/urh #923: LimeSDR the carrier still exists after sending the signal

lx2005Jim, 2022-01-20:
##### Expected Behavior

There should be no RF output after sending an signal.

##### Actual Behavior
The carrier still exists after sending an signal.
(I use a CMU200 to observe the RF output.)

##### Steps To Reproduce

1. Select one recorded signal.
2. Connect LimeSDR.
3. Click on 'Replay signal' button.
4. Click on 'Start' button.
5. The signal is sent properly, but after that the carrier on the frequency still exists even after I close the URH.
6. The LED on the board is green which shows that PLL is still locked after sending the signal.

##### Screenshots

[image]
Just click the start button using the above settings.

##### Platform Specifications

 - OS: Windows10 21H1 19043.1288
 - URH version: 2.9.2
 - Python version: Python 3.10
 - Installed via Windows installer
 - SDR environment: PothosSDR-2021.07.25-vc16-x64
 - LimeSDR-USB FW:4 HW:4 Protocol:1 GW:2.23 Ref Clk: 30.72 MHz

lx2005Jim, 2022-01-20:
similar problems: https://discourse.myriadrf.org/t/unmodulated-carrier-transmitted-after-stopping-gr-pf-model/4099

andynoack, 2022-01-20:
As this is not URH specific, our possibilites to help are rather limited.
I think, this might be related to #916.