jopohl/urh #926: BladeRF / Recording corrupted when sample rate is increased above minimum

Semnodime, 2022-01-27:
##### Expected Behavior
Recording a signal longer at a sample rate higher than 2.5MS/s should not interfere with signal integrity.

##### Actual Behavior
Recording 30 seconds at higher sample rates will result in recordings with insufficient count of samples.

I have recorded the signal via the native bladeRF-cli for reference and can ensure, that the bug is not due to bugs in the firmware of the bladeRF.

As I currently don't have a setup to record a pure sine wave, I cannot determine in which way the signal is corrupted precisely. 

##### Steps To Reproduce
1. Install Linux Mint 20.1 Cinnamon
1. Run `apt update && apt upgrade`
1. Install the latest urh from source
1. Install bladeRF dependencies via `apt install git libusb-1.0-0-dev libtecla-dev help2man`
1. Build the latest libBladeRF and bladeRF-cli from source `cmake -DBUILD_DOCUMENTATION=ON -DENABLE_BACKEND_LIBUSB=ON -DINSTALL_UDEV_RULES=ON -DENABLE_LOG_FILE_INFO=ON -DENABLE_LIBTECLA=ON ../`
1. Plug in the bladeRF into USB3 with SuperSpeed
1. Download the latest bladeRF firmware and flash it
1. Download the latest FPGA image and define it as autoloading
1. Open urh
1. Record noise at 5MS/s-60MS/s sample rate for ~30 seconds
1. See error

##### Screenshots
###### Recording for ~30 seconds with 2.5MS/s
[image]

###### Recording for ~30 seconds with 60MS/s
[image]

##### Platform 
 - OS: Linux Mint 20.1 Cinnamon
 - URH: 2.9.2
 - Python: 3.8.10
 - Installed via git / built from source
 - bladeRF-cli: 1.8.0-git-5a146b2a
 - libBladeRF: 2.4.1-git-5a146b2a
 - bladeRF_fw_latest.img: 2021-01-12 - v2.4.0
 - hostedxA4-latest.rbf: 2021-10-04 - v0.14.0

##### Additional Context
Apparently, urh's libbladeRF is outdated.
```
[INFO::Device.py::start_rx_mode] BladeRF: Starting RX Mode
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO::Device.py::log_retcode] BladeRF-OPEN: Success
```

jopohl, 2022-02-22:
Not sure if that is the root cause but these messages in the log seem suspicious:
```
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
```

jopohl, 2022-03-07:
Do you have "Apply DC correction" checkbox activated? For such high sample rates  the DC correction lags a bit. Turning it off should resolve this issue.

closed 2022-03-07 by jopohl