jopohl/urh #927: Cannot open large IQ file

Semnodime, 2022-01-28:
##### Expected Behavior
URH should load the file progressively.

##### Actual Behavior
URH becomes unresponsive and dies after ~2 minutes.

##### Steps To Reproduce
1. Attempt to load a large complex32s file (~10GiB)

##### Platform Specifications
 - OS: Linux Mint 20.1 Cinnamon
 - URH: 2.9.2
 - Python version: 3.8.10
 - Installed via git / built from source

jopohl, 2022-01-30:
How much RAM do you have installed in your machine? It might also help to disable "Autodetect signals on loading checkbox" under "Edit".

Semnodime, 2022-01-30:
@jopohl RAM: 16GB

Weirdly enough, there's no warning or error in the console both prior and even after the crash.

PR #946 (2022-02-22): get magnitudes in cython for less memory consumption
fix #927
closed 2022-02-22 by jopohl