jopohl/urh #968: Export decoded messages to PCAPNG

jpacov, 2022-03-09:
##### Is your feature request related to a problem?
Exporting decoded messages to a standard output file (PCAPNG) is really useful for further analysis.

##### Describe the solution you'd like
Be able to export decoded messages to PCAPNG.

##### Describe alternatives you've considered
I saw you have a PCAP.py file, but seems not complete.
I propose to use pkg **python-pcapng** (https://python-pcapng.readthedocs.io/en/latest/index.html) which does all the job to save a PCAPNG compliant file really simple.
We could added as a 3rd option in the SaveAs dialog you already have.

If you agree I can make the PR with the changes.

jopohl, 2022-03-10:
Back in the day we implemented PCAP support which worked but hit the limit of that file format as an analysis in Wireshark was not suitable (you had to add your own decoder by submitting a PR to the wireshark foundation or so..). If PCAPNG solves that issue - why not! However, I am hesitant to use new dependencies especially for something as simple as writing a file. If you can create a PR without a new dependency I will be happy to accept it.

jpacov, 2022-03-11:
I'll create the export PCAP functionality without adding a new dependency, as you requested.

Regarding how useful this PCAP file can be:
Me and a bunch of colleagues plan to use your URH to record RF signals, and store them as decoded data in PCAP files.
We'll merge them in Wireshark to other PCAPs created in other mediums (LF for example)
To interpret them in Wireshark, we'll create our Lua dissectors that make all the magic so that Wireshark interprets frames accordingly.

Anyway, the only **open point** I have is: **timestamps**. I see the timestamp is assigned at the moment of processing the message (time.now), and we lose the real timestamp of the signal. Do you know if there is a way to recover that timestamp from the HW (hackrf for example)? Or can we at least use the time.now() at the moment of starting the signal recording instead of later when processing? What do you think?

jopohl, 2022-03-11:
>To interpret them in Wireshark, we'll create our Lua dissectors that make all the magic so that Wireshark interprets frames accordingly.

Good stuff! Just saw [in their wiki](https://wiki.wireshark.org/Lua) how that works, seems like a good route to take, didn't know Wireshark is extensible in that way.

> Anyway, the only open point I have is: timestamps. I see the timestamp is assigned at the moment of processing the message (time.now), and we lose the real timestamp of the signal. Do you know if [...]

Good point...a relative timestamp for a message should be easy to calculate from the sample rate and the location (=current sample) in the signal. Would that be enough i.e. the signal starts at timestamp 000000? If absolute timestamps are required a straightforward solution that comes to mind is to add a timestamp to the `src/urh/signalprocessing/Signal.py` class and set it when an object of that class is created.

jopohl, 2022-03-12:
There should be only one signal created (if we speak about the record dialog). I do not recall at which time the signal is created:
* when the dialog is opened, or
* when the record button is clicked, or
* or when dialog is closed / save button is pressed

should be easy to find this out through debugging. Maybe it makes sense to update/set the timestamp when the record button is pressed.

jpacov, 2022-03-12:
Question: what is the proper way to build repo with HackRf support?
I installed first:
`sudo apt install libhackrf-dev`
And then when installing setup:
`python3 setup.py install`
it says it found it: _Found hackrf lib. Will compile with native hackrf support_

But in interface still only RTL-TCP can be selected in Device.

I guess a flag when building is missing maybe?

jopohl, 2022-03-12:
You need to enable it in Options -> Device.

jopohl, 2022-10-17:
Closed via #970

closed 2022-10-17 by jopohl