jopohl/urh #972: Signal list not scrollable

Apokalypser, 2022-03-24:
##### Expected Behavior
I have six signals opened, but can only see four and a half. Scrolling down enables me to see the other signals.

##### Actual Behavior
I have i.e. six signals opened, but can only see four and a half. Scrolling down the list to see the rest of the signals is not possible.

##### Steps To Reproduce
Open six signals
Try to scroll down

##### Screenshots
[image]

##### Platform Specifications
 - OS: macOS 12.1
 - URH version: 2.9.3
 - Python version: 2.7.18
 - Installed via download of dmg

andynoack, 2022-03-24:
I cannot reproduce this behaviour (just tested with linux). Have you tried a clean reinstallation?

Apokalypser, 2022-03-24:
Yes, just downloaded 2.9.3. Maybe it's a macOS-only bug?

andynoack, 2022-03-24:
Would you try the git version?

Apokalypser, 2022-03-25:
I tried the git version. It supports darkmode now, but the signal list still is not scrollable.

andynoack, 2022-03-25:
That seems to be a macOS bug. Can anyone with a MAC confirm that?

Apokalypser, 2022-04-29:
I can confirm that. After connecting a USB mouse a scrollbar is displayed which enables scrolling. Scrolling on the plain GUI is still not possible. Activating the scrollbar is also possible by setting "Always show scrollbars" in macOS.

closed 2022-05-12 by andynoack