jopohl/urh #975: Does not work with USB3

DBMandrake, 2022-04-24:
Just trying this software for the first time (on a Windows 10 PC) and could not get it to work initially on either my Airspy Mini or RTL-SDR Blog V3 adaptors.

On the Airspy Mini:

"AirSpy-OPEN: Error Code: -5 (-5)
AirSpy: failed to start rx mode"

On the RTL-SDR device:

"RTLSDR-OPEN (#0): Error Code: -1 (-1)
RTLSDR: failed to start rx mode"

After a while I discovered the cause - I have both adaptors plugged into USB 3 ports on my PC for improved performance, (although the devices are USB2, the USB3 controller on the motherboard has better performance at high transfer rates for bulk transfers) and this software does not seem to work with USB 3 controllers, at least on this motherboard. (Gigabyte GA-78LMT-USB3 6.0)

All the other SDR software I use - over a dozen different packages/programs work fine with the devices connected to USB 3 ports, with the exception of very old rtl-sdr command line utilities which have a similar problem, and only work on USB 2 ports.

In the case of those old rtl-sdr command line utilities I was able to fix the issue with USB 3 ports by replacing their outdated libusb-1.0.dll with a more up to date version, then they work fine.

However replacing libusb-1.0.dll alone in my URH installation does not seem to fix this issue so there may be more to it than just this one dll file.

andynoack, 2022-04-26:
I know that this may not be very helpful, but the AirSpy Mini works with USB3 on my PC without problems. Maybe @jopohl has an idea or can reproduce the bug.

jopohl, 2023-10-28:
URH on WIndows ships with the recent-ish [libusb 1.0.25](https://github.com/jopohl/sdrbuild/blob/master/.github/workflows/windows.yml#L45). The utilized AirSpy and RTL-SDR driver versions can be seen [here](https://github.com/jopohl/sdrbuild/releases/tag/v1.1).

Did you try replacing the `rtl-sdr.dll` or `airspy.dll` files as well in URHs installation folder? I wonder whether there are special compiler flags that need to be set to enable USB3 support. I grepped through RTL-SDRs and AirSpys source code but did not fine anything.