jopohl/urh #976: Starting urh.exe throws an exception

DoganM95, 2022-04-25:
##### Expected Behavior
Should start normally

##### Actual Behavior
Starting urh.exe throws an exception

##### Steps To Reproduce
1. Install URH
2. double click urh.exe in install folder

##### Screenshots
[image]

##### Platform Specifications
 - OS: Windows 10
 - URH version: 2.9.3
 - Python version: 3.10.3
 - Installed via GitHub Release (x64 exe)

andynoack, 2022-05-05:
Can someone confirm? I cannot reproduce this error.

closed 2022-05-06 by andynoack

DoganM95, 2022-05-26:
Just fixed this by deleting the `.../Appdata/Roaming/urh` folder.
Seems like my config was messed up or something, just in case anyone runs into this in the future.