jopohl/urh #989: Generate reuasable format from demodulated raw capture data

ResistanceIsUseless, 2022-07-28:
A few tools out there specifically the FlipperZero capture raw rf data as a demodulated number sequences. Would it be possible to add support for importing and or converting these in the generator or Analysis tools? Ideally I'm looking for a way to transfer captures between devices. So it would be cool if you could also export into this format.

Here is an example capture:
```Filetype: Flipper SubGhz RAW File
Version: 1
Frequency: 315000000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: RAW
RAW_Data: 337 -426 363 -888242 167 -356 105 -368 93 -380 327 -126 353 -126 337 -128 339 -128 337 -128 93 -358 347 -132 333 -122 341 -128 121 -370 101 -368 91 -382 317 -134 141 -362 105 -336 127 -356 95 -370 349 -130 329 -124 337 -128 337 -130 123 -3698 97 -374 129 -338 127 -342 351 -140 325 -142 335 -96 345 -126 337 -128 125 -368 341 -140 305 -132 359 -94 121 -374 101 -368 93 -384 351 -102 141 -364 103 -336 129 -372 103 -360 347 -108 361 -106 339 -130 323 -124 123 -3710 131 -360 103 -358 105 -370 327 -142 335 -128 327 -140 361 -106 343 -102 137 -352 353 -94 345 -138 337 -126 97 -376 105 -370 91 -396 331 -132 101 -358 107 -370 93 -394 101 -362 347 -106 363 -106 339 -130 355 -92 121 -3706 129 -342 129 -338 129 -340 347 -124 339 -128 369 -96 337 -128 339 -124 125 -354 347 -132 333 -122 339 -126 121 -372 101 -366 91 -382 351 -102 143 -362 105 -334 129 -356 93 -372 349 -132 329 -124 335 -128 337 -128 125 -3698 131 -360 103 -376 105 -334 353 -140 333 -126 347 -94 369 -96 371 -96 125 -370 329 -140 337 -126 351 -94 123 -372 101 -368 93 -382 351 -104 141 -362 105 -336 127 -358 93 -370 349 -132 329 -124 337 -128 337 -128 125 -3704 97 -392 103 -342 137 -334 353 -138 335 -126 361 -106 359 -106 345 -102 135 -356 357 -106 347 -102 365 -92 121 -374 103 -368 125 -366 331 -132 103 -358 105 -370 93 -394 103 -360 349 -106 361 -106 339 -130 355 -94 121 -3712 133 -358 101 -358 105 -370 363 -106 337 -128 349 -94 369 -96 371 -96 125 -370 361 -108 337 -128 351 -94 121 -372 101 -368 93 -384 351 -102 143 -362 105 -336 127 -372 105 -360 349 -106 361 -108 339 -128 355 -92 123 -3710 131 -358 103 -358 107 -370 329 -140 337 -126 351 -94 369 -96 369 -98 125 -368 363 -108 335 -128 351 -94 121 -374 101 -368 93 -382 351 -104 141 -362 105 -336 127 -374 103 -360 349 -108 361 -106 339 -130 355 -94 121 -3714 99 -392 103 -358 107 -368 327 -140 335 -128 349 -94 391 -104 359 -106 105 -362 357 -106 347 -140 329 -94 139 -342 127 -360 93 -392 327 -122 121 -350 139 -334 127 -356 93 -372 347 -132 331 -124 335 -128 337 -130 123 -3698 133 -358 103 -378 105 -334 353 -140 335 -126 347 -94 369 -96 371 -96 125 -370 361 -108 337 -128 351 -94 121 -372 101 -368 93 -382 351 -104 141 -362 105 -336 127 -358 93 -372 349 -130 331 -124 337 -128 337 -128 125 -3700 129 -340 129 -340 127 -342 343 -126 `

andynoack, 2022-08-12:
Hey, I just committed a branch that allows to load .sub files from the FlipperZero. With my setup it works but shows up a pretty bad signal quality.
Could you check if it works for you?

I just implemented loading but not exporting (at least yet), because URH will drop further information like the frequency directly after loading a sub. Users may not understand why loading and then exporting the same sub file ends up with wrong (i.e. default) signal settings in the file. I am a little bit afraid of (valid) usability complaints...What do you think?

ResistanceIsUseless, 2022-08-16:
This is great. Let me grab it and test. thank you!!

andynoack, 2022-08-17:
I have added a rudimentary export feature for .sub files. All parameters are constants and export will only work for .sub-files.

ResistanceIsUseless, 2022-08-17:
I'm having problems installing the branch manually on my macbook air M1. Seems to be an issue with pyqt5. Going to try on my Linux machine tonight.

ResistanceIsUseless, 2022-08-18:
Got it working on a different machine and it seems fine. I'm honestly not sure what to expect though. I'm able to have enough scale to see the binary/hex data stream. I checked most of my raw captures and they always show as digital type signal. I'm assuming since this is a raw capture it's always going to be an OOK type of modulation or is that just what I've captured? (I honestly don't really understand this raw format besides assuming its a plot on a graph)What did your signal quality look like? I attached some more samples if you wanted to compare.

I see your point about it dropping configuration. You could either:
1. Try and parse the frequency line on line 3 of the raw sub file when importing and prompt the user for anything else missing when saving.
2. Add an alert that lets the user know when saving no signal settings.
3. Add an alert that lets the user know when importing is has no signal settings.
4. Maybe require enabling this format on start. So users understand it's limitations?

[Archive.zip](https://github.com/jopohl/urh/files/9366647/Archive.zip)
d

andynoack, 2022-08-18:
The .sub files contain something like a quadrature demodulated signal whereby there are only two possible amplitudes: 0 and 1. Reading the files as demodulated data was actually not a big problem, so I guess, the RX part is alright.

Saving .sub files is a little bit more complicated. Information is missing and as URH is designed to work with real SDRs, the backend does not really make it easy to work with those files.
In theory of shouldn't be a problem to ask the user for missing information but that would require some dirty proprietary hacks that interferes with URHs generic approach or introduces usability problems.

For a start, an export of demodulated signals to .sub is available - works more or less (you have to edit the file for frequencies != 433M and so on)

Probably the best solution would be to implement a plugin similar to the Yardstick plugin (which should even use the same internal hardware...) to save arbitrary files as .sub-file in the Generator-Tab. This, however, is a lot of work if all the Flipper settings should be supported. I am not sure, if it is worth it to implement it right now because a rudimentary export support is already there.

ResistanceIsUseless, 2022-08-18:
Plugin isn't a bad idea, or just leave it as a branch for now to see how flipper evolves. I originally asked flipper to create a tool to export raw captures as something more flexible for other tools and they declined.

I really like the idea of a plugin though. One of the biggest problems iIface trying to investigate different signals is so no unifying tool unless you're an expert with GNURadio. So adding more to UHR plugins and providing documentation on creating your own would make this feature optional and also easier for someone like me to tinker with.