jopohl/urh #993: Better Documentation for urh_cli

EdwinFairchild, 2022-08-02:
##### Is your feature request related to a problem?

- I keep getting asked for `modulation parameters` but there is no documentation of proper syntax and what are my options. 
- Furthermore I am not modulating, I am only passing the `-rx` parameter and settings things that relate to demodulation so that also has me scratching my head and thinking, what modulation parameters? 
##### Describe the solution you'd like

- Just better documentation of the cli interface in general. Some features  of the GUI are also undocumented and found them through someone else's question and answer to themselves. 
- ascii files filled with ones and zeros can get huge, an option for binary output of the captures would be great.
##### Describe alternatives you've considered

##### Additional context

TomKranenburg, 2023-04-27:
+1 for this. Trying to decipher the command line arguments is very taxing. The only example given is out of date. Making the CLI interface redundant.

TomKranenburg, 2023-04-27:
Example, this:

  -e ENCODING, --encoding ENCODING
                        Specify encoding
                        
 What options are available?

TomKranenburg, 2023-04-28:
To any poor soul drifting through this unknown code. There are some examples in here:

https://github.com/jopohl/urh/blob/71153babe497a1dafa12f8dd371ed0adc305a8bb/tests/cli/test_cli_parsing.py