jopohl/urh #994: URH with X310 and Twin RX

alphafox02, 2022-08-04:
##### Expected Behavior
Select supported sample rate of 50 or 100msps

##### Actual Behavior]
Double Free or Corruption shown in terminal windows upon starting spec a 

[INFO::Device.py::log_retcode] USRP-OPEN (type=x300,addr=192.168.40.2,fpga=HG,name=,serial=31A0C7B,product=X310): Success
[INFO::Device.py::log_retcode] USRP-SET_SUBDEVICE to : Success
[INFO::Device.py::log_retcode] USRP-SET_ANTENNA_INDEX to 0: Success
[INFO::Device.py::log_retcode] USRP-SET_FREQUENCY to 433.92M: Success
[INFO::Device.py::log_retcode] USRP-SET_SAMPLE_RATE to 50M: Success
[INFO::Device.py::log_retcode] USRP-SET_BANDWIDTH to 50M: Success
[INFO::Device.py::log_retcode] USRP-SET_RF_GAIN to 0.25: Success
Odouble free or corruption (out)

##### Steps To Reproduce

Start URH 2.9.3, select spec a, attempt to start with 50M or 100M in Sample rate/bandwidth. Although bandwidth is limited I think to 80MHz wide per channel on the Twin RX.

##### Platform Specifications
Ubuntu 20.04 (DragonOS) w/ UHD 3.15

Happy to test further while I have this device available. Although, I guess it wouldn't be of much use using such a large sample rate/bandwidth in URH?

andynoack, 2022-08-04:
Certainly it depends on your use case whether 50M or 100M sample rate is required, we usually work with 1M or 2M.
The double free error (or memory corruption) has to be in our cython extension or the UHD library. When setting up the device we don't call free() and raise a memory errors when you are out of RAM, so my first guess is that the error maybe out of the URH scope. Have you tried to recompile the cython extensions? Maybe by manually installing URH from source?

alphafox02, 2022-08-04:
I’ll double check, but in my case I built URH this way. 

git clone https://github.com/jopohl/urh/
cd urh
python setup.py install

I’ll pay closer attention this evening and test everything I can think of. I figured while I have the device temporarily that I’d check as many open source applications as I can in all sorts of different ways.

andynoack, 2023-01-25:
closed due to inactivity

closed 2023-01-25 by andynoack

alphafox02, 2023-01-28:
I should’ve mentioned, I still have an x310 if further testing/feedback would be helpful.