jopohl/urh #1005: unable to build with sdrplay >= 3.0

ZeroChaos-, 2022-09-29:
##### Expected Behavior
it builds successfully

##### Actual Behavior
```
INFO: compile options: '-I/usr/include/python3.10 -I/usr/lib/python3.10/site-packages/numpy/core/include -c'
extra options: '-Wno-cpp'
INFO: x86_64-pc-linux-gnu-gcc: src/urh/dev/native/lib/sdrplay.cpp
src/urh/dev/native/lib/sdrplay.cpp:778:10: fatal error: mirsdrapi-rsp.h: No such file or directory
  778 | #include "mirsdrapi-rsp.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

##### Steps To Reproduce
Install sdrplay 3.0 or higher (I am using 3.07.1)
try to build --with-sdrplay

##### Platform Specifications
 - OS: Gentoo Linux
 - URH version: 2.9.3 and git HEAD 6ae113d64f2c0da39911dbe41f9e374f5c3e4e2a
 - Python version: 3.10
 - Installed via gentoo package manager (I am the package maintainer for gentoo)

jopohl, 2022-10-17:
URH does not support SDRPlay API >= 3.0 yet, see https://github.com/jopohl/urh/issues/783

ZeroChaos-, 2022-11-09:
duplicate of https://github.com/jopohl/urh/issues/783