jopohl/urh #1014: RSP1A not connecting

BlackHawk2212, 2022-11-01:
Hi everyone, 
I've just downloaded and installed Universal Radio Hacker and I've tried follwing some tutorials to use URH.
Unfortunately, it's not connecting to my RSP1A.
I get an error like this:
`
SDRPlay: CONNECTED DEVICES: 
SDRPlay-SET DEVICE NUMBER to 0: HARDWARE 
ERROR (7)
`
Can anyone sort me out??

Thanks!

andynoack, 2022-11-01:
Talking about the SDRPlay, there is only native support for v2.13 API. Have you tried the GNU Radio backend?

BlackHawk2212, 2022-11-01:
Hi @andynoack,
I've installed GNURadio and put the python interpreter directory but the same this occurs when I try it again. Here is a screenshot of the options menu if it helps...
[image]
The error message is the same from before...

Kind Regards and thank you for your time!
BlackHawk

andynoack, 2022-11-01:
It seems that GNU Radio is not available in your installation. Unless URH does not indicate that it can use your GNU Radio installation, the backend will not be used and thus the errors message does not change. Maybe something with your PATH is wrong or the installed version is not compatible with URH (should better be a Python2 based version).
We cannot give full installation support here. Please consider using the Slack channel if the ideas don't help. Maybe trying out URH in Linux is also a good way to go.

BlackHawk2212, 2022-11-01:
Right. Thank you.Is using GNU Radio the only way to go?What are the other options to solving the original problem rather of no connection between the RSP1A??Thanks!

andynoack, 2022-11-01:
I would give linux a try. It is the platform URH was developed on and it is usually more verbose with errors.

BlackHawk2212, 2022-11-02:
Hi Andy,
I can't install Linux at my institution...not even on a VM. 
Can we try anything else??

andynoack, 2022-11-02:
I do not know whether you have a clean installation of Windows and all the libs and drivers. So it is hard to say what the problem is.
This is, however, not a technical bug in URH. Please use our Slack channel for community support. I will close here.

closed 2022-11-02 by andynoack