jopohl/urh #1015: Generating fuzzing hangs a long time

FalcoGer, 2022-11-10:
##### Expected Behavior
Data is generated quickly, ui becomes responsive in a reasonable time.

##### Actual Behavior
UI hangs exceedingly long after fuzzing. The more bits are fuzzed, the longer things hang up.

##### Steps To Reproduce

1. Go to Generator
2. Right click on message list
3. Select create empty message, 12 bits
4. Select all 12 bits
5. Right click, add label
6. Generate fuzzing output (4095)
7. Click label in label list
8. Click Fuzz
9. Watch the list fill instantly, but then the UI hangs for a long time.

##### Screenshots

##### Platform Specifications

 - OS: Ubuntu Mate 22.04
 - URH version: 2.9.3
 - Python version: 3.10
 - Installed via pip

FalcoGer, 2022-11-12:
I think I might know why. I am using a custom encoder/decoder written in python. I think it runs all the modulation in the background after it generated the fuzzing output. Because it's calling python thousands of times, it just doesn't respond while it's doing that.

andynoack, 2022-11-12:
Hey @FalcoGer,

I think, you are pretty much right with your assumption on the external decoding plugin. You can, however, avoid it if you just want to preprend something, e.g. a single bit.. Just right click in Generator->"Generated Data" and "Insert column"->"on the left". Then edit the contents of the added column (number 1) as you like it to be. After that continue with fuzzing of this modified line using fuzzing labels.
This might solve your problem! Can we close this issue?

FalcoGer, 2022-11-12:
Yes this can be closed, although an option in the decoders window would be nicer, since I'd like to type my data into the generator, not individual bis. The protocol I'm trying to generate for uses a series of bits for 1, a different series of bits for 0 and has some special bit strings here and there that are always the same (for example the 1 bit at the start, or if the data is shorter than max there are some special bits to replace the missing data)

closed 2022-11-12 by FalcoGer