jopohl/urh #1019: rad1o initialisation fails because of missing bias-tee

9R, 2023-01-12:
##### Expected Behavior
Use rad1o (a hackRF compatible badge) as sdr in urh spectrum analyser

##### Actual Behavior
Divice initialisation fails with
```
[ERROR::Device.py::log_retcode] HackRF-SET_BIAS_TEE_ENABLED to 0: HACKRF_ERROR_LIBUSB (-1000)
```
Reason: rad1o does not have a software controllable bias tee

I think this issue could resolved in various ways:
 *  Checkbox in the hackRF profile to let urh know there is a rad1o in use which completely disables the bias-tee initialization 
 *  An additional device profile for the rad1o without the bias-tee code
 *  Catch the libusb error without closing the connection

Completel log:
```
[INFO::Device.py::start_rx_mode] HackRF: Starting RX Mode
[INFO::Device.py::log_retcode] HackRF-SETUP (0000000000000000750865d9222da693): Success
[INFO::Device.py::log_retcode] HackRF-SET_FREQUENCY to 433.92M: Success
[INFO::Device.py::log_retcode] HackRF-SET_SAMPLE_RATE to 2M: Success
[INFO::Device.py::log_retcode] HackRF-SET_BANDWIDTH to 2M: Success
[INFO::Device.py::log_retcode] HackRF-SET_RF_GAIN to 0: Success
[INFO::Device.py::log_retcode] HackRF-SET_IF_GAIN to 16: Success
[INFO::Device.py::log_retcode] HackRF-SET_BB_GAIN to 14: Success
[ERROR::Device.py::log_retcode] HackRF-SET_BIAS_TEE_ENABLED to 0: HACKRF_ERROR_LIBUSB (-1000)
[INFO::Device.py::stop_rx_mode] HackRF: Stopping RX Mode: Stop on error
[INFO::Device.py::log_retcode] HackRF-Start RX MODE: Success
[INFO::Device.py::log_retcode] HackRF-STOP RX MODE: Success
[INFO::Device.py::log_retcode] HackRF-CLOSE: Success
[INFO::Device.py::log_retcode] HackRF-EXIT: Success
```

##### Steps To Reproduce

1. Open Spectrum Analyser
2. Select Device: HackRK
3. Update device ID
4. Press Start
5. HackRF not found popup  appears

##### Screenshots

##### Platform Specifications

 - OS: Arch Linux
 - URH version: 2.9.3
 - Python version: 3.10
 - Installed via pacman
 - rad1o firmware version: REV 05 | 2022-10-26 | Update to upstream HackRF version v2022.09.1

9R, 2023-01-12:
I just ran a short test to confirm my theory:
After disabling the bias_tee lines in dev/config.py and dev/native/HackRF.py when running urh from source the rad1o works as expected.

andynoack, 2023-01-13:
Hey @9R, would you try if the new branch rad1o_support works for you?

9R, 2023-01-13:
@andynoack Thank you for the quick response. I just ran a short test with spectrum analyzer, recorder and protocol sniffer and the rad1o works as expected with your changes in the new rad10_support branch.

PR #1020 (2023-01-13): Add rad1o support
Add support for rad1o SDR (like HackRF but without bias_tee)
(fix #1019)
closed 2023-01-13 by andynoack