jopohl/urh #1032: Failed to Execute Script 'main' recording a signal

Gliebig, 2023-02-10:
##### Expected Behavior
I have been recording individual signals trying to decode a TPMS. I had 10 tests recorded in a new project and all was good. But the last test was to simulate an air loss so I started recoding the signals and slightly unscrewed the sensor from the valve stem. I had about 6-8 pulses recorded and then the program crashed.

##### Actual Behavior
Error window: Failed to execute script 'main' due to unhandled exception: 'NoneType' object has no attribute 'write'
Traceback (most recent call last):
  File "multiprocessing\process.py", line 297, in _bootstrap
  File "multiprocessing\process.py", line 99, in run
  File "urh\dev\native\Device.py", line 176, in device_receive
  File "urh\dev\native\RTLSDR.py", line 46, in receive_sync
  File "multiprocessing\connection.py", line 200, in send_bytes
  File "multiprocessing\connection.py", line 290, in _send_bytes
BrokenPipeError: [WinError 109] The pipe has been ended

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "urh\main.py", line 187, in <module>
  File "PyInstaller\hooks\rthooks\pyi_rth_multiprocessing.py", line 49, in _freeze_support
  File "multiprocessing\spawn.py", line 105, in spawn_main
  File "multiprocessing\spawn.py", line 118, in _main
  File "multiprocessing\process.py", line 312, in _bootstrap
AttributeError: 'NoneType' object has no attribute 'write'

##### Steps To Reproduce

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

##### Screenshots

##### Platform Specifications

 - OS: Windows 10 Pro ver 22H2 OS Build 19045.2486
 - URH version: 2.94
 - Python version: [e.g. 3.7]
 - Installed via downloaded executable

andynoack, 2023-02-10:
It looks like the connection to your RTLSDR broke down (BrokenPipeError). After that URH cannot receive any information from the SDR and returns an empty object (NoneType) to the URH GUI that also produces some error messages (but shouldn't completely crash) then. 
Please ensure that you are using latest (at least stable) RTLSDR drivers. Maybe recompile the cython extensions by calling the build.py in ../cythonext or using the device menu.

Gliebig, 2023-02-11:
Thank you for your reply. The multiple messages were not saved and that explains that. The drivers are current as I just installed the software and drivers last week when I received my SDR. I'm just stuck trying to decode the data I have now. Some of it makes sense and other bits do not.

andynoack, 2023-02-11:
I am afraid, we cannot do much about that as the error did not occur within the URH code. So there might be something wrong with your setup.

Please try to
- reproduce the error (to see if it is a persistent error)
- rebuild the cython Extensions (to fix interface problems)
- run URH under linux (to exclude driver issues)

andynoack, 2023-02-15:
Are there any new insights based on my comment? I would like to close the issue due to inactivity soon.

andynoack, 2023-03-08:
closed due to inactivity

closed 2023-03-08 by andynoack