jopohl/urh #1033: URH 2.9.4 does not recognise SDRs

kalle-vaan, 2023-02-20:
##### URH 2.9.4 should recognise SDRs

##### The software replies the following:

USRP205: 
USRP-OPEN: Error Code: 11 (11)
USRP: Could not retrieve more detailed error message from device.
USRP: failed to start rx mode

SDRPlay:

SDRPlay: CONNECTED DEVICES: 
SDRPlay-SET DEVICE NUMBER to 0: HARDWARE ERROR (7)

and opens the following window: Unhandled exception in script
X; Failded to execute script 'main' due to unhandled exception:
'NoneType' object has no attribute 'write' 

Traceback (most recent call last):
  File "multiprocessing\process.py", line 297, in _bootstrap
  File "multiprocessing\process.py", line 99, in run
  File "urh\dev\native\Device.py", line 146, in device_receive
  File "urh\dev\native\SDRPlay.py", line 93, in init_device
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "urh\main.py", line 187, in <module>
  File "PyInstaller\hooks\rthooks\pyi_rth_multiprocessing.py", line 49, in _freeze_support
  File "multiprocessing\spawn.py", line 105, in spawn_main
  File "multiprocessing\spawn.py", line 118, in _main
  File "multiprocessing\process.py", line 312, in _bootstrap
AttributeError: 'NoneType' object has no attribute 'write'

##### Platform Specifications

 - OS: Windows 11
 - URH version: 2.9.4 also 2.9.3

andynoack, 2023-02-20:
Maybe this is related to #1032
Seems to be a driver problem under Windows which may not be in the URH code (that throws an excecption). Can you try it under linux? Please have a look in the other issue as well, there are already some suggestions which may also fit here.

kalle-vaan, 2023-02-20:
> Lähetetty: maanantai 20. helmikuuta 2023 11.07
> Vastaanottaja: ***@***.***>
> Kopio: ***@***.***>; ***@***.***>

I’ll check the other issues. And with Linnux later on. However the two years old UHR version worked perfectly. I’m not sure about the driver problem because with other software, like SDRAngel, both radios are recognised.
Kalle

Lähetetty Windowsin Sähköposti<https://go.microsoft.com/fwlink/?LinkId=550986>ista

npbk21, 2023-03-02:
> ##### URH 2.9.4 sẽ nhận ra SDR
> ##### Phần mềm trả lời như sau:
> USRP205: USRP-OPEN: Mã lỗi: 11 (11) USRP: Không thể truy xuất thông báo lỗi chi tiết hơn từ thiết bị. USRP: không khởi động được chế độ rx
> 
> SDRPlay:
> 
> SDRPlay: THIẾT BỊ ĐƯỢC KẾT NỐI: SỐ THIẾT BỊ SDRPlay-SET thành 0: LỖI PHẦN CỨNG (7)
> 
> và mở cửa sổ sau: Ngoại lệ chưa được xử lý trong tập lệnh X; Không thể thực thi tập lệnh 'chính' do ngoại lệ chưa được xử lý: đối tượng 'NoneType' không có thuộc tính 'ghi'
> 
> Traceback (lần gọi gần đây nhất): Tệp "multiprocessing\ process.py", dòng 297, trong _bootstrap Tệp "multiprocessing\ process.py", dòng 99, trong tệp run "urh\dev\native\Device.py", dòng 146, trong tệp device_receive "urh\dev\native\SDRPlay.py", dòng 93, trong init_device IndexError: danh sách chỉ mục nằm ngoài phạm vi
> 
> Trong quá trình xử lý ngoại lệ trên, một ngoại lệ khác đã xảy ra:
> 
> Traceback (lần gọi gần đây nhất): Tệp "urh\main.py", dòng 187, trong Tệp "PyInstaller\hooks\rthooks\pyi_rth_multiprocessing.py", dòng 49, trong Tệp _freeze_support "multiprocessing\spawn.py", dòng 105 , trong Tệp spawn_main "multiprocessing\spawn.py", dòng 118, trong Tệp _main "multiprocessing\ process.py", dòng 312, trong _bootstrap AttributeError: đối tượng 'NoneType' không có thuộc tính 'write'
> 
> ##### Thông số nền tảng
> * Hệ điều hành: Windows 11
> * Phiên bản URH: 2.9.4 cũng như 2.9.3

Try it.
https://www.rtl-sdr.com/rtl-sdr-quick-start-guide/

nop0x900, 2023-03-28:
> ##### URH 2.9.4 should recognise SDRs
> ##### The software replies the following:
> USRP205: USRP-OPEN: Error Code: 11 (11) USRP: Could not retrieve more detailed error message from device. USRP: failed to start rx mode
> 
> SDRPlay:
> 
> SDRPlay: CONNECTED DEVICES: SDRPlay-SET DEVICE NUMBER to 0: HARDWARE ERROR (7)
> 
> and opens the following window: Unhandled exception in script X; Failded to execute script 'main' due to unhandled exception: 'NoneType' object has no attribute 'write'
> 
> Traceback (most recent call last): File "multiprocessing\process.py", line 297, in _bootstrap File "multiprocessing\process.py", line 99, in run File "urh\dev\native\Device.py", line 146, in device_receive File "urh\dev\native\SDRPlay.py", line 93, in init_device IndexError: list index out of range
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "urh\main.py", line 187, in File "PyInstaller\hooks\rthooks\pyi_rth_multiprocessing.py", line 49, in _freeze_support File "multiprocessing\spawn.py", line 105, in spawn_main File "multiprocessing\spawn.py", line 118, in _main File "multiprocessing\process.py", line 312, in _bootstrap AttributeError: 'NoneType' object has no attribute 'write'
> 
> ##### Platform Specifications
> * OS: Windows 11
> * URH version: 2.9.4 also 2.9.3

For SDRPlay RSP1.
To solve this issue, I installed SDRPlay API 2.13.1, be sure to reboot the system and plug the device !!!

kalle-vaan, 2023-04-14:
> Lähetetty: tiistai 28. maaliskuuta 2023 19.56
> Vastaanottaja: jopohl/urh ***@***.***>
> Kopio: kalle-vaan ***@***.***>; Author ***@***.***>

Thank for the advice. After installing API Hw/driver 3.09 SDRPlay RSP2Pro seems to work properly. API 3.09 is the second last version.

Kind regards
Kalle

closed 2023-06-02 by andynoack