jopohl/urh #1042: RSP1 - the signal is not recorded correctly (incorrect Sample rate)

nop0x900, 2023-03-28:
##### Expected Behavior

RSP1 device. If you set the sample rate to 2M and the bandwidth to 125k (250k), then the real sample rate after recording is 500k. And the record length is exactly 4 times less. Well, accordingly, the width of the pulses is also 4 times smaller. With a bandwidth of 500k, the real sample rate is 1M (2M is specified in the settings).

##### Actual Behavior

##### Steps To Reproduce

Sample rate 2M, Band width 125k,250k -> real sample rate is 500k
Sample rate 2M, Band width 500k -> real sample rate is 1M

##### Screenshots

![scr1](https://user-images.githubusercontent.com/60018319/228217174-abb773a4-e15d-4818-ae46-fe004e4ba765.pn
[image]
g)

##### Platform Specifications

Windows 7 x86

andynoack, 2023-03-28:
The bandwidth does not influence the sample rate.

nop0x900, 2023-03-28:
> Полоса пропускания не влияет на частоту дискретизации.

> The bandwidth does not influence the sample rate.

Yes. But there is a bug...

andynoack, 2023-04-07:
I am unsure what exactly the problem is. By the way, you can configure the sample rate of a signal with a click on the (i) button. This is, however, only for displaying purposes (including the calculation of the shown timings).

nop0x900, 2023-04-08:
> I am unsure what exactly the problem is. By the way, you can configure the sample rate of a signal with a click on the (i) button. This is, however, only for displaying purposes (including the calculation of the shown timings).

Thank you, that's exactly what I do. Only here is the problem with playback, through hack rf (minimal sample rate 2M), and it turns out it does not work correctly. And it's not convenient, you have to set the parameters every time through the (i) button.

andynoack, 2023-04-08:
Well, playback/replay should not work at all using different sample rates for recording and sending, as the data stream is passed to the sending device directly.
However, if you generate this signal, it is possible. I understand your problem as follows: 
- you record something with SDRplay and sample rate n 
- URH does not set the sample rate you had configured
- bandwidth does not matter as long as you deactivated the lock in the recording dialog. If not, you are going to change the sample rate and this issue has no point.

andynoack, 2023-04-20:
Closed due to inactivity.

closed 2023-04-20 by andynoack