jopohl/urh #1043: Only RTL-TCP Shows Up

muhuna, 2023-04-01:
I have an RTL-SDR device and in the past used it with URH on a different system. I moved to a new system and can't get URH to recognize my RTL-SDR device. When selecting the source I only see RTL-TCP. The dropdown menu does not show any other option.

[image]

I did install AirSpy and that program does show RTL-SDR USB. I can also listen in to FM radio so I know it's working. 

[image]

Someone had a similar issue and deleted the uhr.ini file but in my case that didn't help (shut down URH, delete ini file, restart URH).

I run on Windows 10 and under device manager it shows up as Universal Serial Bus devices - RTL2838UHIDIR (again, it works under AirSpy).

I have Python 2.7, 3.8 and 3.9 installed. UHR runs under 3.9. URH is on v2.9.4 and was installed via pip.

andynoack, 2023-04-03:
Please activate your SDR actives under Edit->Options->Device.

closed 2023-04-03 by andynoack

andynoack, 2023-04-03:
This occurs when the development libraries for your system (e.g. sudo apt install libhackrf-dev) are not installed on your system. After you have installed them, URH is going to make the devices available.