jopohl/urh #1044: No binary output from urh_cli

internetrelative, 2023-04-06:
##### Expected Behavior
Expected the demodulated ASK signal to output a binary which is to be written to the file

I'm able to receive raw using:
urh_cli -v -d  RTL-SDR -f 433920000 -s 1000000 -fcorr 1  -rx  -pm 0 --receive-time -1  --raw --file output.complex16s

##### Actual Behavior
[INFO::settings.py::get_receive_buffer_size] Allocate receive buffer with 400MB
[INFO::Device.py::start_rx_mode] RTLSDR: Starting RX Mode
Receiving forever...
Found Rafael Micro R820T tuner
[INFO::Device.py::log_retcode] RTLSDR-OPEN (#0): Success
Exact sample rate is: 1000000.026491 Hz
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY to 433.92M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_SAMPLE_RATE to 1M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_BANDWIDTH to 1M: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_FREQUENCY_CORRECTION to 1: Success
Disabled direct sampling mode
[INFO::Device.py::log_retcode] RTLSDR-SET_DIRECT_SAMPLING_MODE to 0: Success
[INFO::Device.py::log_retcode] RTLSDR-SET_RF_GAIN to 14: Success
[INFO::Device.py::log_retcode] RTLSDR-RESET_BUFFER: Success

##### Steps To Reproduce
urh_cli -v -d  RTL-SDR -f 433920000 -s 1000000 -fcorr 1  -rx  --parameters 0.0,100.0 -bps 1  -c 0.12114343047142029  -cs 1 -mo ASK -sps 300 -t 3 -rt -1 --file output.dat

What does parameters actually do? Most of these values are copied from the URHProject.xml
##### Platform Specifications

 - OS: Raspbian GNU/Linux 11 (bullseye) armv7l
 - URH version: [e.g. 2.9.4]
 - Python version: [e.g. 3.9.2]
 - Installed via [installled from source]

andynoack, 2023-04-06:
I see to command lines without any errors and with no further information. You need to provide helpful information and a specific error description to get help.

internetrelative, 2023-04-07:
There was no output from  urh_cli -v -d RTL-SDR -f 433920000 -s 1000000 -fcorr 1 -rx --parameters 0.0,100.0 -bps 1 -c 0.12114343047142029 -cs 1 -mo ASK -sps 300 -t 3 -rt -1 --file output.dat, I had use --raw and manually copy and paste the .complex16 into URH to see the binary data.

andynoack, 2023-04-07:
-cs and -bps are not required here. Try also: --parameters 0 1.

internetrelative, 2023-04-07:
Still no output with it, only the file gets created. urh_cli -v -d RTL-SDR -f 433920000 -s 1000000 -fcorr 1 -rx --parameters 0 1 -c 0.12114343047142029  -mo ASK -sps 300 -t 3 -rt 5 -g 49 --file output.dat  
Here's how it looks like when I manually imported the raw .complex16s 
[image]

andynoack, 2023-04-07:
Try smaller values for parameters, e.g. "0.1 0.7" and I guess, -c 0.1265 should be precise enough. This is, however, an individual fine tuning issue and not a general error in URH so lets move to Slack. You can close here.

closed 2023-04-07 by internetrelative