jopohl/urh #1045: WSP checksum8 calculation ignores the last byte

vagnerjazon, 2023-04-07:
##### Expected Behavior
The checksum should be calculated from every byte after the preamble and sync word until just before the checksum byte.

##### Actual Behavior
The last byte before the checksum is not included in the calculation.

##### Steps To Reproduce
1. Set the checksum type to WSP, either to automatic or to forced checksum8
2. The calculated checksum is incorrect, the differrence (mod 0xff) is exactly the value of the last byte before the checksum itself

##### Platform Specifications
 - OS: Windows 10
 - URH version: 2.9.4
 - Python version: 3.11.3
 - Installed via: installer from "Releases"

I was wondering if I should create a pull request instead, since I have found the problem in WSPChecksum.py, but I have never created one, so I just write the reason here:
In WSPChecksum.py, the checksum8 function already receives the message without the last byte, so in line 102 "len(bits) - 8" should be just "len(bits)".

andynoack, 2023-04-07:
I just reviewed this line of code. We need to prevent access on unallocated memory that might happen if len(bits) is not a multiple of 8. With this in mind, we cannot just omit the mentioned "-8" for i which is later used as [i:i+8]. The code must be (memory) safe in general.

Please re-check the settings of data in CRC settings, maybe by tweaking this values you can already solve the problem. Otherwise we need a fix that does not increase the crash risk for len(bits)%8!=0.

vagnerjazon, 2023-04-08:
Indeed, I did not think about that possibility.
Although the checksum4 function does exactly that and for those kind of packets it works fine for me.
https://github.com/jopohl/urh/blob/f52c0730346d0887500ae87e3151a00327d60c52/src/urh/util/WSPChecksum.py#L94-L95
Regarding the checksum8 function, the change could instead be at the calling side in the same file in lines 57 and 62, because if the entire "msg" was passed on to the function instead of "msg[:-8]", that would also solve the problem.
https://github.com/jopohl/urh/blob/f52c0730346d0887500ae87e3151a00327d60c52/src/urh/util/WSPChecksum.py#L57
https://github.com/jopohl/urh/blob/f52c0730346d0887500ae87e3151a00327d60c52/src/urh/util/WSPChecksum.py#L62
If the entire msg was passed on, then the "ignore trailing hash" would be done by the "len(bits) - 8" part in the checksum8 function.
I have tested this by installing from source on Ubuntu and it worked as I expected, the calculated checksums were correct.

The crc8 works fine as it is now, as there are no further data truncation after the initial msg[:-8].

I cannot tweak the checksum settings for the WSP checksum, there is no such data range setting possibility as for the CRC, I can only select either Automatic or "Force Checksum8".

andynoack, 2023-04-08:
The best approach is probably to implement a test-driven fix. Can you provide me with maybe two examples in the format you see in test_crc.py? 

I think the problem occurs with not byte aligned data, because when only one bit is missing, a complete byte would be skipped. The code must have a solution for this and must not read from unused memory (i.e. array positions) in edge cases. In general I prefer skipping several bits to become byte aligned over beeing creative about missing/trailing bits...

vagnerjazon, 2023-04-08:
I did not find checksum testing in test_crc.py, but found a test for this here:
https://github.com/jopohl/urh/blob/f52c0730346d0887500ae87e3151a00327d60c52/tests/test_encoding.py#L126-L133
Both testcases use the WSPChecksum object in "ChecksumMode.auto" mode (since this is the default and the test does not change that). This means that the hash type is automatically selected, and because both test messages have CRC as hash (the MSB of STATUS is 1), the checksum8() is not tested.
https://github.com/jopohl/urh/blob/f52c0730346d0887500ae87e3151a00327d60c52/src/urh/util/WSPChecksum.py#L40-L43
So I propose adding a third test case, where that bit is 0.
For by example slightly modifying msg1 (the last byte is 0x40 instead of 0x80):
`msg3 = util.hex2bit("a6d201006401009802019e411e40b6")`
`checksum3 = util.hex2bit("b6")`
`calc_checksum3 = wsp_checker.calculate(msg3)`
`self.assertTrue(calc_checksum3 == checksum3)`

I predict that currently the calculated checksum will be 0x76 instead of 0xb6.

andynoack, 2023-04-08:
Should be fixed now in master branch. Please close the issue after verifying :-)

vagnerjazon, 2023-04-08:
It works now as expected.
Thank you. :)

closed 2023-04-08 by vagnerjazon