jopohl/urh #1047: Crashes when trying to use spectrum analyzer. i got the newest version of urh. i have x64 win10

hhjjj75, 2023-04-12:
Traceback (most recent call last):
  File "multiprocessing\process.py", line 297, in _bootstrap
  File "multiprocessing\process.py", line 99, in run
  File "urh\dev\native\Device.py", line 177, in device_receive
  File "multiprocessing\connection.py", line 257, in poll
  File "multiprocessing\connection.py", line 328, in _poll
BrokenPipeError: [WinError 109] Putki on lopetettu

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "urh\main.py", line 187, in <module>
  File "PyInstaller\hooks\rthooks\pyi_rth_multiprocessing.py", line 49, in _freeze_support
  File "multiprocessing\spawn.py", line 105, in spawn_main
  File "multiprocessing\spawn.py", line 118, in _main
  File "multiprocessing\process.py", line 312, in _bootstrap
AttributeError: 'NoneType' object has no attribute 'write'
[image]

andynoack, 2023-04-13:
Please try to
- reproduce the error (to see if it is a persistent error)
- rebuild the cython Extensions (to fix interface problems)
- run URH under linux (to exclude driver issues)

andynoack, 2023-04-18:
Closed due to inactivity.

closed 2023-04-18 by andynoack