jopohl/urh #1050: urh does not run in linux mint 20.3

geckobob, 2023-04-27:
Installed via pip. "urh --version" gets "2.9.4". 
Running urh with "urh" gets the following:
Traceback (most recent call last):
  File "/usr/local/bin/urh", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/urh/main.py", line 84, in main
    import urh.cythonext.signal_functions
  File "src/urh/cythonext/signal_functions.pyx", line 1, in init urh.cythonext.signal_functions
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 80 from PyObject

I'm not a programmer. No idea what to do with this.

andynoack, 2023-04-28:
Please try the URH version from git. There seems to be a version collision with Python/Cython somehow. By the way, the current Mint version is 21.1 and it probably uses a later Python version.

sysfu, 2023-04-29:
getting the same issue on Elementary OS 7 Horus, will try the git version.

sysfu, 2023-04-29:
Git install worked, no problems.

andynoack, 2023-07-12:
Closed due to inactivity

closed 2023-07-12 by andynoack